realthunder/FreeCAD #203: question (not an issue) - order of things in tree view

bambukouk, 2021-11-08:
Is it possible to re-arrange the order of items listed in the tree view?
Functionally it doesn't make any difference to FreeCAD, but to me as a user keeping things tidy and in (my preferred) order makes things easy to handle.
Unfortunately depending on the order I create things, they end up listed a bit randomly (from my human/user point of view)...

I know that in Assembly 3 one can arrow things up and down easily - is this (or something similar) possible in link branch tree view outside of Assembly 3? 
or... is it still down to main branch and it's limitations?

wohltat, 2021-11-13:
I really miss sorting elements in tree view. Optimally this should be as simple as dragging it to another position with a line indicating where it will be placed.

closed 2021-12-05 by realthunder

bambukouk, 2021-12-05:
Thank you @realthunder 
Hopefully it will not suffer from the problems described on FreeCAD forum :+1:

wohltat, 2021-12-05:
awesome, looking forward to see this in action

commit 638f3bbf (2021-12-05): App/Gui: support object reordering through drag and drop

Refactored from upstream commit 345dcd52529ffe907e9312da557d135f38f1dfe2

Root level item is ordered by new property 'TreeRank' in
DocumentObject. It is added to DocumentObject instead of ViewObject to
support undo/redo.

New APIs added to help reordering objects, Document::reorderObjects(),
Document::treeRanks() (for tree rank ranges, exposed to Python as
Document.TreeRanks). User can also reorder objects by directly modify
TreeRank property.

Children items reordering is done by new API
ViewProvider::canReplaceObject()/replaceObjects( [... cut]