realthunder/FreeCAD #211: Spreadsheet workbench -- color picker disappears after switching to different workbench

scottmudge, 2021-11-27:
Perhaps this is a known issue, the color picker elements of the Spreadsheet Workbench are only visible the first time the workbench is activated:

[image]

But when switching to a different workbench (e.g., Part Design), and then switching back to the Spreadsheet Workbench, the color picker elements disappear:

[image]

I briefly looked into Workbench.cpp of the spreadsheet module, but nothing immediately seemed wrong. Beyond the color pickers being the only dynamically allocated UI elements (created during the initialization of the Spreadsheet workbench).

closed 2021-11-28 by realthunder

scottmudge, 2021-11-28:
So was this due to the local lambda expression being used as the event handler? I thought that might have been it, and I see you moved the handler to a dedicated function, but there are a few other changes as well.

realthunder, 2021-11-28:
That's not the problem. The Spreadsheet toolbar uses non-standard way of adding those color picker widget to toolbar. The FreeCAD ToolBarManager will treat them as unused and remove them every time the Spreadsheet workbench is activated.

If you check the `SpreadsheetGui::Workbench::activated()` function in upstream, you'll find it has an variable `initialized` but never set it to true. This bug causes it to add those two pickers everytime the workbench is activated, which happens to counter the fact that they will be removed by ToolBarManager on activation.

scottmudge, 2021-11-29:
I see, thank you for taking the time to explain, and for debugging the issue so quickly!

commit 60fcc6a5 (2021-11-28): Spreadsheet: fix color pick disappearing in toolbar

Fixes #211
@scottmudge