realthunder/FreeCAD #212: How to build bgfx lib?

KuzemkoA, 2021-12-01:
I try build LinkDaily branch on windows 10. If I understand correct, link don't found bgfx lib. Are you build it separate? 
`
  [1/2248] Linking CXX shared library src\Gui\Renderer\FreeCADRenderer.dll
  FAILED: src/Gui/Renderer/FreeCADRenderer.dll src/Gui/Renderer/FreeCADRenderer.lib 
  cmd.exe /C "cd . && "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_dll --intdir=src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1430~1.307\bin\Hostx64\x64\link.exe /nologo src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir\FreeCADRenderer_autogen\mocs_compilation.cpp.obj src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir\Renderer.cpp.obj  /out:src\Gui\Renderer\FreeCADRenderer.dll /implib:src\Gui\Renderer\FreeCADRenderer.lib /pdb:src\Gui\Renderer\FreeCADRenderer.pdb /dll /version:0.0 /machine:x64 /NODEFAULTLIB /debug /INCREMENTAL -LIBPATH:C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5OpenGLd.lib  C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Widgetsd.lib  C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Guid.lib  C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Cored.lib  kernel32.lib user32.lib gdi32.lib winspool.lib SHFolder.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib winmm.lib comsupp.lib Ws2_32.lib dbghelp.lib  && cd ."
  LINK Pass 1: command "C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1430~1.307\bin\Hostx64\x64\link.exe /nologo src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir\FreeCADRenderer_autogen\mocs_compilation.cpp.obj src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir\Renderer.cpp.obj /out:src\Gui\Renderer\FreeCADRenderer.dll /implib:src\Gui\Renderer\FreeCADRenderer.lib /pdb:src\Gui\Renderer\FreeCADRenderer.pdb /dll /version:0.0 /machine:x64 /NODEFAULTLIB /debug /INCREMENTAL -LIBPATH:C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5OpenGLd.lib C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Widgetsd.lib C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Guid.lib C:\FreeCADLibs_asm3_12.5.5_x64_VC17\lib\Qt5Cored.lib kernel32.lib user32.lib gdi32.lib winspool.lib SHFolder.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib winmm.lib comsupp.lib Ws2_32.lib dbghelp.lib /MANIFEST /MANIFESTFILE:src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir/intermediate.manifest src\Gui\Renderer\CMakeFiles\FreeCADRenderer.dir/manifest.res" failed (exit code 1120) with the following output:
     Creating library src\Gui\Renderer\FreeCADRenderer.lib and object src\Gui\Renderer\FreeCADRenderer.exp
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol "void * __cdecl operator new(unsigned __int64)" (??2@YAPEAX_K@Z) referenced in function "public: static void * __cdecl std::_Default_allocate_traits::_Allocate(unsigned __int64)" (?_Allocate@_Default_allocate_traits@std@@SAPEAX_K@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol "void __cdecl operator delete(void *,unsigned __int64)" (??3@YAXPEAX_K@Z) referenced in function "void __cdecl std::_Deallocate<16,0>(void *,unsigned __int64)" (??$_Deallocate@$0BA@$0A@@std@@YAXPEAX_K@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol atexit referenced in function "void __cdecl `dynamic initializer for '_ResourcePath''(void)" (??__E_ResourcePath@@YAXXZ)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol __imp__invalid_parameter referenced in function "void * __cdecl std::_Allocate_manually_vector_aligned<struct std::_Default_allocate_traits>(unsigned __int64)" (??$_Allocate_manually_vector_aligned@U_Default_allocate_traits@std@@@std@@YAPEAX_K@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol __imp__CrtDbgReport referenced in function "void * __cdecl std::_Allocate_manually_vector_aligned<struct std::_Default_allocate_traits>(unsigned __int64)" (??$_Allocate_manually_vector_aligned@U_Default_allocate_traits@std@@@std@@YAPEAX_K@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl std::_Lockit::_Lockit(int)" (__imp_??0_Lockit@std@@QEAA@H@Z) referenced in function "private: void __cdecl std::_Iterator_base12::_Adopt_locked(struct std::_Container_base12 const *)" (?_Adopt_locked@_Iterator_base12@std@@AEAAXPEBU_Container_base12@2@@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl std::_Lockit::~_Lockit(void)" (__imp_??1_Lockit@std@@QEAA@XZ) referenced in function "private: void __cdecl std::_Iterator_base12::_Adopt_locked(struct std::_Container_base12 const *)" (?_Adopt_locked@_Iterator_base12@std@@AEAAXPEBU_Container_base12@2@@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol __std_exception_copy referenced in function "public: __cdecl std::exception::exception(class std::exception const &)" (??0exception@std@@QEAA@AEBV01@@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol __std_exception_destroy referenced in function "public: virtual __cdecl std::exception::~exception(void)" (??1exception@std@@UEAA@XZ)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol memcpy referenced in function "private: void __cdecl std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >::_Memcpy_val_from(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?_Memcpy_val_from@?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEAAXAEBV12@@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol memmove referenced in function "class Render::RendererLib * * __cdecl std::_Copy_memmove<class Render::RendererLib * *,class Render::RendererLib * *>(class Render::RendererLib * *,class Render::RendererLib * *,class Render::RendererLib * *)" (??$_Copy_memmove@PEAPEAVRendererLib@Render@@PEAPEAV12@@std@@YAPEAPEAVRendererLib@Render@@PEAPEAV12@00@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol "void __cdecl std::_Xlength_error(char const *)" (?_Xlength_error@std@@YAXPEBD@Z) referenced in function "void __cdecl std::_Throw_tree_length_error(void)" (?_Throw_tree_length_error@std@@YAXXZ)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol _CxxThrowException referenced in function __catch$??$_Emplace_reallocate@AEBQEAVRendererLib@Render@@@?$vector@PEAVRendererLib@Render@@V?$allocator@PEAVRendererLib@Render@@@std@@@std@@QEAAPEAPEAVRendererLib@Render@@QEAPEAV23@AEBQEAV23@@Z$0
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2019: unresolved external symbol _RTC_CheckStackVars referenced in function "public: static class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > __cdecl Render::RendererFactory::types(void)" (?types@RendererFactory@Render@@SA?AV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@XZ)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2001: unresolved external symbol _RTC_InitBase
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2001: unresolved external symbol _RTC_Shutdown
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2001: unresolved external symbol __CxxFrameHandler4
[... cut ...] ned __int64)" (?compare@?$_Narrow_char_traits@DH@std@@SAHQEBD0_K@Z)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\Renderer.cpp.obj : error LNK2001: unresolved external symbol "const type_info::`vftable'" (??_7type_info@@6B@)
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\LINK : error LNK2001: unresolved external symbol _DllMainCRTStartup
C:\Users\kuzem\Source\Repos\FreeCAD\out\build\x64-Debug\src\Gui\Renderer\FreeCADRenderer.dll : fatal error LNK1120: 20 unresolved externals
`

realthunder, 2021-12-02:
The above commit fixes the build for Windows.

By default the new renderer module is just a skeleton that does not include `bgfx`, because it is not ready for public testing yet. I only tried it on Linux with a simple demo. I plan to work on it more after the pending new release.

`bgfx` is included as a git submodule. In the future, you'll need to run `git submodule update --recursive` to get the source code first. And then reconfigure FreeCAD cmake with `BUILD_BGFX=On` to build the renderer with `bgfx`.

luzpaz, 2022-01-18:
Closing ticket

closed 2022-01-18 by luzpaz