realthunder/FreeCAD #215: step file export of faces generated by Surface WB

wsteffe, 2021-12-04:
Hello RT,
  You have already introduced a "Togge Export" command which can be applied to faces generated by Part/Estrude.
Would it be possible to extend it also to faces generated by the Surface WB ? I need to use this WB to fill areas enclosed by sketches.

[test10.zip](https://github.com/realthunder/FreeCAD/files/7653716/test10.zip)

Another problem is that the faces generated by the Surface WB are not created inside the active Body.
In the annexed document the Surface object was generated (by the Surface WB) when the body named BC_STRIP was active.
But it doesn't appear inside of the BC_STRIP body.

The situation is different for the Extrude001 object which was generated (by the Part WB) when the Body named SPLIT was active.
In fact Extrude001 is inside of the SPLIT Body.

A third (minor problem) is that color settings do not work with surface objects. The command "Set color" can be used but it doesn't have any effect.

realthunder, 2021-12-05:
The PartDesign has the `Wrap` feature to incorporate feature from external workbench into PartDesign body. The `Extrude001` feature in your file is wrapped before adding to body. Currently the wrapping is only possible if the feature include dependency on only features inside the current active body. The above reference commit adds the possibility to create `Wrap` on exiting feature use drag and drop.

Using your file as an example, you can activate body `DIEL_CAVITY` first, and then create filling surface. The moment you select the sketch edge, a dialog will prompt asking whether you want to wrap the feature into the active body.

[image]

In case you somehow rejects the operation, and make it remember your choice, the dialog will not show up in the future. To bring it back, you can go to `Preference -> Part design` as shown below.

[image]

realthunder, 2021-12-05:
And about the color, how do you set the color? Right click the object in the tree view, and choose 'Set colors...'. I just tried and it is working.

wsteffe, 2021-12-06:
Thanks, as it can be seen in the here annexed example, now the the Surface feature can be wrapped inside of a Part Design Body. [test11.zip](https://github.com/realthunder/FreeCAD/files/7660320/test11.zip)

You said "Currently the wrapping is only possible if the feature include dependency on only features inside the current active body. "
But it seems to me that this limitation is not required. In my example both Sketch001 and Sketch002 include an external reference to an edge that belongs to the body DIEL_SUBS1. Sketch001 is used by Surface001 which is wrapped in BC_STRIP. Sketch002 is used by Extrude001 which is wrapped in SPLIT_1.

Regarding the color setting it seems to me that it works with Surface001 (which is a single face) but not with Extrude001 which has generated two separate faces.

wsteffe, 2021-12-15:
I have just checked again and I have seen that now colors setting work also for set of faces (compounds).

closed 2021-12-15 by wsteffe