realthunder/FreeCAD #216: [Minor] Missing icon `TreeDragReorder.svg`

scottmudge, 2021-12-05:
It will probably be added soon, but just making a note that `TreeDragReorder.svg` is missing but required by the latest commits to `LinkDaily`:

```
RCC: Error in 'C:/Development/3D/FreeCAD/src/Gui/Icons/resource.qrc': Cannot find file 'TreeDragReorder.svg'
```

wsteffe, 2021-12-05:
Yes, I have the some problem: 
```
Building ViewProviderPy.h/.cpp out of ViewProviderPy.xml
make[2]: *** No rule to make target '../src/Gui/Icons/TreeDragReorder.svg', needed by 'src/Gui/qrc_resource.cpp'.  Stop.

```

scottmudge, 2021-12-05:
For the time being it's possible to copy and paste an existing icon to fill the role of the missing icon.

realthunder, 2021-12-05:
Fixed. Thanks for the heads up.

commit 32b9b569 (2021-12-05): Gui: add missing icon

Closes #216
closed 2021-12-05 by realthunder