realthunder/FreeCAD #217: Keyboard shortcuts for QAction objects inside TreeWidget are not working

scottmudge, 2021-12-08:
I'm not sure if this is due to event filters not properly passing keystrokes though or what, but I've noticed that the `->setShortcut()` bindings in the `TreeWidget::TreeWidget()` constructor are not being triggered.

The only one set is F2 for rename, which obviously works because it is an innate function of the Tree view (and properly redundant to set), but if you try to set it to a different key, it still only responds to F2.

Furthermore, I am trying to add a shortcut key to the `hideInTreeAction` and `showHiddenAction` actions (Ctrl+H and Ctrl+Shift+H or some variant on that), as I use these features quite frequently, but using `setShortcut()` does nothing. The shortcuts are never triggered, regardless of which key is set.

scottmudge, 2021-12-08:
I forgot to make a note that the file of interest is `Tree.cpp`, though you have probably figured this out.

realthunder, 2021-12-09:
It's because those actions are never added to `TreeWidget` (via `QWidget::addAction()`), they are only added to context menu which is created on demand. Normally for this kind of functionality, you'll create a FreeCAD `Command` so the user can manage the shortcut and so on. I suggest you create new commands for those and group them under [Std_TreeViewActions](https://github.com/realthunder/FreeCAD/blob/effc472978dadb433c80fdbc2cd81a81433728a3/src/Gui/CommandView.cpp#L3821) command.

scottmudge, 2021-12-09:
I see, so the `Qt::F2_Key` binding in `Tree.cpp` is purely visual to let the user know the binding exists, and not necessarily functional. I will test this.

closed 2021-12-09 by scottmudge

scottmudge, 2021-12-09:
But wait... what is the point then of Qt even providing the keyboard shortcut feature for context menu items? Is it just broken for this purpose? Is the only way to actually get a keyboard shortcut working in FreeCAD to create a new global action in `CommandView.cpp`?

I guess that kind of makes sense in order to keep things coherent in terms of keyboard accessibility.

What would your suggested method of binding "Hide item" and "Show hidden items" to new commands under `Std_TreeViewActions` without creating duplicate code?

scottmudge, 2021-12-09:
Nevermind, I think I've got it figured out. I'm guessing the standard `setShortcut()` method is just to add keybindings to the standard menu-item type entries.

It does make sense to create global commands for this instead.

I have added the new commands and needed code here:

https://github.com/realthunder/FreeCAD/compare/LinkDaily...scottmudge:Feature_TreeHideShowCommands?expand=1

Does it look okay? Would these new commands be valuable to merge into your fork or the main repo?