realthunder/FreeCAD #219: building with OCC 7.6.0

wsteffe, 2021-12-09:
Hello RT,
 I wanted to switch to OCC 7.6.0 but I have found a few problems which made me change the following files:
-    src/Mod/Part/App/Geometry.cpp
-    src/Mod/Part/App/TopoShapeEx.cpp

Here annexed are the diff:
[Diff.zip](https://github.com/realthunder/FreeCAD/files/7687305/Diff.zip)

The removal of  BRepAdaptor_HCurve and similar, ... and the need to replace them with  BRepAdaptor_Curve ..
is documented https://dev.opencascade.org/doc/overview/html/occt__upgrade.html

The change made with IsPlanar() and IsLinear() are due to the removal of Surface() and Curve() from classes Adaptor3d_Surface  and Adaptor3d_Curve. The changed functions are compatible with OCC7.6 but are not exactly the same as before. I am just checking the Surface/Curve types.

To make a real planarity/linearity check within a maximum deflection it would be necessary to convert Adaptor3d_Surface to Geom_Surface and BRepAdaptor_Curve to Geom_Curve. Perhaps using the following functions declared in GeomAdaptor.hxx:

- Standard_EXPORT static Handle(Geom_Curve) MakeCurve (const Adaptor3d_Curve& C);
- Standard_EXPORT static Handle(Geom_Surface) MakeSurface (const Adaptor3d_Surface& theS,
                                                           const Standard_Boolean theTrimFlag = Standard_True);

wsteffe, 2021-12-09:
The FreeCAD build against OCC 7.6.0 has just finished now. 
So I may confirm that the only required changes (for compatibility with OCC 7.6.0) are those reported above.

wsteffe, 2021-12-09:
I have changed again the file src/Mod/Part/App/Geometry.cpp to have exactly same behaviour as before. Here annexed are the new diff with respect to repository:

[Diff2.zip](https://github.com/realthunder/FreeCAD/files/7688455/Diff2.zip)

wsteffe, 2021-12-10:
I have tried to make a push request into LinkDaily but it always goes into a wrong place.
First trial went into FreeCAD master. Second tone went into another place (I don't have even understood where it has finished).

realthunder, 2021-12-10:
Thanks for the fix. I took your patch and made a few changes to make it work with older OCC as well.

commit 83d634c3 (2021-12-10): Part: fix build for OCC 7.6

Fixes #219
closed 2021-12-10 by realthunder