realthunder/FreeCAD #221: [Request] Parameterize/expression-ize boolean properties

scottmudge, 2021-12-10:
There are a few boolean properties I would like to be able to control/parameterize in a spreadsheet (or base on other variables/properties), which I currently have to manually toggle. For instanace:

[image]

How difficult would it be to expand the expression functionality to treat zero results (epsilon of 1.0×10−5 or something) as `false`, and non-zero results as `true`, and then to bind this result to boolean properties, like in the above screenshot?

Is this something that you've thought about adding?

realthunder, 2021-12-11:
Just right click the property name and choose `Expression...`. Float can be implicitly converted to boolean just like in Python. Or, you can use any conditional expression.

scottmudge, 2021-12-11:
I see, I guess I did not click on the property name when attempting to right click before. It seems like if the property value combo box is active or selected and you right click, the context menu fails to appear. But that's probably just a quirk of Windows.

Thanks!

scottmudge, 2021-12-15:
Actually, I am re-opening this issue, because the `Model Thread` boolean toggle is the **only** one which lacks the `Right Click > Expression` option from the context/right-click menu. 

`Threaded` can be set, but because this modifies the diameter of the hole, I would prefer to modify `Model Thread`, as it would allow me to improve recalculation performance by disabling it in my spreadsheet until I want to export the final model.

realthunder, 2021-12-23:
Because `ModelThread` is set to `ReadOnly` in case `Threaded` is `False`. You can manually clear the `ReadOnly` by turning on `Show all` in property view context menu. Once you've cleared the flag, you need to de-select the object and re-select again to enable the `Expression...` action.

scottmudge, 2021-12-23:
I see, thank you for looking into that.

scottmudge, 2021-12-24:
Verified resolved through `Show all` workaround.

closed 2021-12-24 by scottmudge