realthunder/FreeCAD #222: Issue with the AliasManager macro and and the spreadsheets in LinkStage3 Daily branch

VorpalBlade, 2021-12-12:
I'm not sure if this is a LinkStage3 bug or a bug in the AliasManager macro.

The issue is that text cells seems to have an extra ' at the beginning of the strings. These are added automatically, and are readded if I try to get rid of them. This can also be seen in the GUI:
[image]

The AliasManager macro reads this, and tries to assign that string as the alias for the cell next to it. A leading ' is not valid in an alias name however, so the macro fails. This works fine on 0.19, and used to work on LinkStage3 before (a few months ago?)

If this is the intended behaviour (why?) I'll report a bug to the AliasManager macro (at https://github.com/pgilfernandez/FreeCAD_AliasManager). It would be easy enough to work around with a `.lstrip("'")` call. However, this behaviour seems really strange, so I assume it is not intended.

---
Version info copied from the help dialog:
```
OS: Arch Linux (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

By the way, that locale in the help dialog seems wrong/over-simplified, as it only considers LC_MESSAGES:
```console
~ ❯ locale                      
LANG=sv_SE.UTF-8
LC_CTYPE="sv_SE.UTF-8"
LC_NUMERIC="sv_SE.UTF-8"
LC_TIME="sv_SE.UTF-8"
LC_COLLATE="sv_SE.UTF-8"
LC_MONETARY="sv_SE.UTF-8"
LC_MESSAGES=en_US.UTF-8
LC_PAPER="sv_SE.UTF-8"
LC_NAME="sv_SE.UTF-8"
LC_ADDRESS="sv_SE.UTF-8"
LC_TELEPHONE="sv_SE.UTF-8"
LC_MEASUREMENT="sv_SE.UTF-8"
LC_IDENTIFICATION="sv_SE.UTF-8"
LC_ALL=
```

realthunder, 2021-12-23:
The problem should be fixed by 40d34595347aacd53bcac8e05295d2ea89ca4e39

VorpalBlade, 2022-01-01:
Awesome. I'll test once a new Daily AppImage is available.

luzpaz, 2022-01-18:
@VorpalBlade pls retest on latest Daily, thx

closed 2022-01-19 by luzpaz

VorpalBlade, 2022-01-19:
Issue is fixed with latest version