realthunder/FreeCAD #223: Regression: failing polar pattern

wsteffe, 2021-12-13:
On my LinkDaily FC (below more details on my build) the here annexed simple document fails to compute: [test12.zip](https://github.com/realthunder/FreeCAD/files/7703404/test12.zip). The failing feature is the last one: the PolarPattern.

OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.30668 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 7edee4b30be4e74c2d38b560f19caf52ba114e8f
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.6.0
Locale: English/United States (en_US)

wsteffe, 2021-12-14:
I have just seen that the problem occurs only if the number of occurrences is set to 1 (which anyway is the default value).

wsteffe, 2021-12-14:
I was wrong in my last statement. The problem occurs anyway but it is only a wrong error message displayed in the report view: **Recompute failed! Please check report view.**

The polar pattern is computed and it seems ok.

wsteffe, 2021-12-14:
I have just understood that the problem is in effect related with occurrences=1. It happens also in other cases because initially this number is set to 1 (the default value). 
Changing the angle (while the occurrences is set to 1) forces an automatic recomputing (because of the activated show preview). This computation fails producing the message "Recompute failed! Please check report view".
After increasing the number of occurrences (let say to 2) the pattern is computed correctly but the error message is still there in the report view. From the documentation it seems that the angular spacing should be (specified angle)/(number of occurrences-1). So it is clear that the computation must give an error when the number is 1. If it is so It would be better to set to 2 the default and to add a validator to force this number to be always >1.

wohltat, 2021-12-14:
Your project fails even with occurences > 1.
If i delete the polar pattern and recreate it i see the behaviour you described.

I dont think forcing the occurences value > 1 is a good idea. It should also work with a value of "1" to avoid problems in parametric designs.

If i check `New shape` the `occurences = 1` works without an error message. This would explain why there are errors with non-contiguous objects, but i don't understand why there is an error with single part objects.

wsteffe, 2021-12-14:
From the documentation it seems that the number of occurrences includes the initial one (the base shape). So when you specify number of occurrences=1 nothing has to be done because the base shape is already there. This is a special condition but yes, it could simply return without giving an error.

commit 19275216 (2021-12-15): PartDesign: fix pattern feature with single occurence

Fixes #223
@wsteffe
closed 2021-12-15 by realthunder