realthunder/FreeCAD #224: Sketcher: creating externally linked edges of circles and arcs has stopped working

xplwowi, 2021-12-14:
To recreate the problem, attached [test.zip](https://github.com/realthunder/FreeCAD/files/7707898/test.zip) can be used:

1. Create new sketch on XZ_Plane
2. Import circular or arched edges from two solids
3. Reference lines will not projected correctly onto the sketch plane

Environment:
OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.30668 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 7edee4b30be4e74c2d38b560f19caf52ba114e8f
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Polish/Poland (pl_PL)

xplwowi, 2021-12-14:
A screenshot of an attempt to import all edges
[image]

wsteffe, 2021-12-15:
Hello xplwowi, are you still having these error messages ?
About a weak ago I also had error messages when importing edges from a sketch.
It was a strange behaviour: first trial gave an error and second one were successful.
But now (with the last sync) I do not have anymore this problem.

I have just tried your test case and I have no problem importing circular edges. See annexed images where imported edges are the two orange.
[image]

xplwowi, 2021-12-15:
Hi westeffe. Thanks for verification but the problem is still there. I just compiled LinkDaily with the patches from the last days.
You showed the projection to the XY_Plane, which works fine for me as well. The problem appears when projecting perpendicular - to the XZ_Plane. This is how I showed it in the screenshot above. The projection of the top edges of solids should result in two segments located above the X axis. I broke the complicated model on which I work. Fortunately I found a temporary solution - vertex projection still works correctly and one endpoint can be imported from a circle or two from an arc.

That's how it has worked so far and it still works in upstream version:
[image]

xplwowi, 2021-12-15:
The same scene but wrong projection under LinkDaily:
[image]

wsteffe, 2021-12-15:
I see. you are right. There is a problem with projection ox XZ plane.
The following screenshot taken from my build shows  two projections:
1) The top linear edge (a diameter) of the half cylinder. This is OK
2) The top circular edge of the full cylinder. This is the one which triggered the error message.  In my opinion it has the correct shape but a bad placement.

[image]
.

wsteffe, 2021-12-15:
To better understand this problem I have done a second test with following steps:
1) Opening the test document provided by xplwowi  and start a sketch in XZ plane
2) Activating the external geometry import
3) Selecting the upper circular edge of the full cylinder:
     Nothing becomes visible after this first click.
4) After selecting again the upper circular edge of the full cylinder the following messages are displayed in the report view

> 18:14:14  Link to Edge3 already exists in this sketch.
> 18:14:14  App.getDocument('test').getObject('Sketch002').addExternal(Part.importExternalObject((FreeCAD.getDocument('test').getObject('Body'), 'Pad001.Edge3'), (FreeCAD.getDocument('test').getObject('Body'), 'Sketch002.')), False)
> 18:14:14  Failed to add external geometry: Not able to add external shape element

...and the imported edge appears in the sketch but with a wrong placement as shown in the following image:

[image]

xplwowi, 2021-12-15:
The problem clearly concerns the projection of a circle and an arc onto a segment. It seems that it appeared with commit https://github.com/realthunder/FreeCAD/commit/343a0f9f6f5b3d0b55b56e74bc59cfb54dcaf46a 

Even slightly tilting the plane makes projection ok, involving "to ellipse conversion".
[image]

commit b7c5f386 (2021-12-16): Sketcher: fix circular external edge projection

Fixes #224
@xplwowi