realthunder/FreeCAD #225: Python3.10.1 Errors

VincentFrangi, 2021-12-14:
I just upgrade my system and got now Python 3.10.1

After recompiling FreeCAD LinkDaily got theses error in the report view on editing my projects:
```python
10:08:00  Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Draft/draftviewproviders/view_base.py", line 290, in onChanged
    if vobj.RootNode.getChild(2).getChildren().getLength() > 0:
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 41024, in getChild
    return _coin.SoGroup_getChild(self, index)
<class 'SystemError'>: <built-in function SoGroup_getChild> returned a result with an exception set
10:08:00  Traceback (most recent call last):
  File "<string>", line 427, in Activated
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1637, in show
    self.setTrackers()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1685, in setTrackers
    self.tracker = trackers.snapTracker()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py", line 154, in __init__
    color.rgb = FreeCADGui.draftToolBar.getDefaultColor("snap")
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3998, in __setattr__
    field = self.getField(name)
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3873, in getField
    return _coin.SoFieldContainer_getField(self, name)

10:08:00  <built-in function SoFieldContainer_getField> returned a result with an exception set
```

Here when I try the draft line tool:

```python
10:15:16  Running the Python command 'Draft_Line' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_lines.py", line 69, in Activated
    super(Line, self).Activated(name)
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_base_original.py", line 287, in Activated
    super(Creator, self).Activated(name, noplanesetup)
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_base_original.py", line 152, in Activated
    Gui.Snapper.setTrackers()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1685, in setTrackers
    self.tracker = trackers.snapTracker()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py", line 154, in __init__
    color.rgb = FreeCADGui.draftToolBar.getDefaultColor("snap")
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3998, in __setattr__
    field = self.getField(name)
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3873, in getField
    return _coin.SoFieldContainer_getField(self, name)

<built-in function SoFieldContainer_getField> returned a result with an exception set
```

FreeCAD
```
OS: Manjaro Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.20.30668 (Git)
Build type: Release
Branch: makepkg
Hash: 7edee4b30be4e74c2d38b560f19caf52ba114e8f
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

Thanks,
Vincent

realthunder, 2021-12-23:
Not exactly sure about the cause, but maybe you can try to recompile pivy the Coin3D python binding.

MicPec, 2021-12-27:
Hi, same here on Archlinux when I try to open Draft WB

```
09:46:12  Traceback (most recent call last):
  File "<string>", line 147, in Activated
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1637, in show
    self.setTrackers()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_snapper.py", line 1681, in setTrackers
    self.grid = trackers.gridTracker()
  File "/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py", line 987, in __init__
    pick.style.setValue(coin.SoPickStyle.UNPICKABLE)
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3989, in __getattr__
    field = self.getField(name)
  File "/usr/lib/python3.10/site-packages/pivy/coin.py", line 3873, in getField
    return _coin.SoFieldContainer_getField(self, name)

09:46:12  <built-in function SoFieldContainer_getField> returned a result with an exception set
```

Draft wb not working at all.
Coin3D recompiled from Yours git, pivy compiled from official git.
 
Freecad
```
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.30728 (Git)
Build type: Release
Branch: makepkg
Hash: f952471510afa085537918569f6298ba3fceb565
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Polish/Poland (pl_PL)
```

realthunder, 2022-01-01:
Looks like FreeCAD is not ready for Python 3.10 yet, mostly due to lack of support in third-party libraries. See [here](https://forum.freecadweb.org/viewtopic.php?style=5&f=15&t=64573).

VincentFrangi, 2022-01-03:
Exactly, it is related to this issue coin3d/pivy#88
I hope someone will take care of it, because lot of arch and fedora users seems unable to use FreeCad now

akshimassar, 2022-01-28:
There is a snap package for @realthunder fork as well as two packages for mainstream version. https://snapcraft.io/freecad-realthunder

It took me about 5 minutes to install freecad from here (I'm arch linux user) and it works great.

luzpaz, 2022-05-30:
Privy 0.6.7 supports python 3.10
https://github.com/coin3d/pivy/issues/88#issuecomment-1140696850