realthunder/FreeCAD #228: Sketcher: wrong arc and circle parallel-to-plane projection

xplwowi, 2021-12-16:
Corrections to issue #224 caused another problem - wrong parallel projections with body-to-body references.

This time problem is harder to avoid. Attached [test2.zip](https://github.com/realthunder/FreeCAD/files/7726362/test2.zip) sample shows it on two schetches:
Sketch "WrongProjection" is supported by bottom plane of upper body - arcs and circles are out of the body, arc has malformed shape.
[image]

Sketch "GoodProjection" is supported by top plane of upper body - import worked ok.
[image]

wsteffe, 2021-12-16:
Hello xplwowi

Annexed patch fixed the problem for me.
I have just reinserted the two lines removed by previous fix but in a different position so that they are executed only for a parallel projections and not for orthogonal projections. 

[SketchObject_diff.txt](https://github.com/realthunder/FreeCAD/files/7726773/SketchObject_diff.txt)

wsteffe, 2021-12-17:
Hello RT,
 the problem with last commit is that for parallel projection case inside following brackets:
```
7220: if (fabs(circle.Radius() - minorRadius) < Precision::Confusion()) {
...
}
```

  the projected circle is not converted into sketch coordinates.
I think that there is also an inconsistency in the argument of:

`7236:  getParameterRange(hCircle, firstProjPt, y, firstParam, lastParam);`

because firstProjPt and lastProjPt was already moved to sketch coordinates at line 7191 while hCircle was defined in global coord.
Temporarily I have fixed that problem by reintroducing, only for the parallel projection, the two lines:
```
cnt.Transform(mov);
circle.Transform(mov);
```
But I do not like this solution because cnt and circle have a wider scope.

realthunder, 2021-12-17:
Thanks for the tip.

xplwowi, 2021-12-17:
I don't want to open a new issue and I don't know if there are any notifications to closed issues. I will comment without conviction whether it will reach people interested. The case also concerns a functionality that does not exist in the upstream.

One, small problem is still there but doesn't affect any of my real models. Arc may be reversed after projecting to the tilted surface. Sample: [test3.zip](https://github.com/realthunder/FreeCAD/files/7736857/test3.zip)

[image]

wsteffe, 2021-12-18:
Hello xplwowi, I can not see anything wrong in your test3:

[image]
[image]

wsteffe, 2021-12-18:
But perhaps the fact that I do not see any arc reversal is because I am using OCC 7.6.

xplwowi, 2021-12-19:
Hi wsteffe. My fault - I have mistakenly attached archive with unsaved, not tilted model. 
[test3.zip](https://github.com/realthunder/FreeCAD/files/7740140/test3.zip) should show the issue.

wsteffe, 2021-12-19:
@RT This problem can be fixed with the annexed patc
[SketchObject_diff.txt](https://github.com/realthunder/FreeCAD/files/7740516/SketchObject_diff.txt)

The line
`7366:  getParameterRange(curve, firstProjPt, lastProjPt, firstParam, lastParam);`

is replaced with the following two lines which exchange firstParam and lastParam for opposite orientations of sketch and circle:

```
 if(vec1.Dot(vec2)>0) getParameterRange(curve, firstProjPt, lastProjPt, firstParam, lastParam);
 else                 getParameterRange(curve, lastProjPt, firstProjPt, firstParam, lastParam);
```

here is a screenshot of
[image]

realthunder, 2021-12-19:
> is replaced with the following two lines which exchange firstParam and lastParam for opposite orientations of sketch and circle:

Wow, this is a much more elegant solution. Thanks!

xplwowi, 2021-12-19:
Hi. I'm afraid to write :)
Changes from above made parallel projection of arc wrong (reversed) again. For tilted planes everything looks fine.
To reproduce:

1. Open document from https://github.com/realthunder/FreeCAD/files/7740140/test3.zip again.
2. Set attachment offset angle to 0° under TopPlate>Pad001>Sketch001
3. Look at arc direction in WrongProjection sketch - it is upside down

wsteffe, 2021-12-19:
The problem now is that the swap:
```
7287           if(vec1.Dot(vec2)<0)
7289                  std::swap(firstProjPt, lastProjPt);
```
applies also to the parallel projection while it should be used only in the oblique projection (after current line 7455).

realthunder, 2021-12-20:
> Hi. I'm afraid to write :)

You testing is highly appreciated, which is why I leave this issue open.

realthunder, 2021-12-20:
> The problem now is that the swap:
> applies also to the parallel projection while it should be used only in the oblique projection (after current line 7455).

By right, as long as the  angle between the circle and the sketch plane exceeds 90 degree, it indicates that we are projecting from the other side. It shouldn't matter whether we are doing oblique projection or not. The reason of the failure of this method is because the sketcher code creates the projected arc differently for the two cases. It transforms the original circle to the sketch plane in case of parallel projection. Since the code is not written by me, not sure if it will change again. I have reverted to the more complex three-point projection to determine the arc orientation, which is agnostic to how the projected geometry is generated.

wsteffe, 2021-12-20:
Yes RT you are right. It is because the code is written differently for the two cases. In fact this code is very confusing: there are a lot of cases sub-cases with a separate algorithm for each one of them. For this reason I was  addressing separately the issues arising in each different situation. The last one discovered by xplwowi was related to the oblique case and then I changed only the code of this case. 
Anyway I think it is safe to use my simpler solution because firstProjPt, lastProjPt are used
only in three places. Two of these places (parallel circle projection and orthogonal circle projection) were OK after my last patch. Eventually we could have to fix the last one related with ellipse projection.

wsteffe, 2021-12-20:
RT, I think that an agnostic solution may not work because you can't decide if you have to swap the endpoints if you don't know how the projected geometry is generated.

realthunder, 2021-12-20:
The three-point method takes the original and the projected curves as inputs and let OCC to project the points, so it does not care how the curve is generated, assuming OCC can do the right thing of course.

Yeah, that projection code is indeed confusing. There might still be cases not handled right there, one of which is the projection of an arc to a line.

xplwowi, 2021-12-20:
I have not found any combinations that cause problems with projecting circles and arcs onto planes. I tried positive and negative tilt angles, projections to faces, planes of the coordinate system, datum planes.
I think I can close the issue. 
Thank you for your support.

closed 2021-12-20 by xplwowi

wsteffe, 2021-12-20:
OK I see, your solution is also correct and will work. But the simpler one (not agnostic) is also safe and easy to handle. After all there are just two types of algorithms:
1) That one used in the parallel projection of a circle arc. In this case the projected arc is obtained from a transformation of the original trimmed curve. Any swap implied by an opposite reparametrization is handled internally by OCC and must not be applied again.

2) That one used in the oblique projection of a circle arc. In this case the projected ellipse is recreated from skretch (with following lines) and you have to take into account of sketch versus circle orientations. 
```
gp_Ax2 refFrameEllipse(gp_Pnt(gp_XYZ(p[0], p[1], p[2])), gp_Vec(0, 0, 1), vecMajorAxis); 
Handle(Geom_Ellipse) curve = new Geom_Ellipse(refFrameEllipse, circle.Radius(), minorRadius);
```

commit b527359e (2021-12-17): Sketcher: fix circular external geometry parallel projection

Fixes #228