realthunder/FreeCAD #229: Not esplicative error message

wsteffe, 2021-12-16:
In a report view I am seeing the following message:
```
10:39:34  A3 available
10:41:06  <Import> ImportOCAF2.cpp(1277): CMP_KaCh2_SFILT_6_4_tunable#Body004 has null shape
```
But I have renamed the Bodies and I am not able to find which one was originally named Body004.
In the Property View associated with a Body there is only the new name (stored in the Label field) but there isn't any indication about the original name.
It would be better if the error message were referring to the user name (instead of using the initial unique name). Otherwise it could refer to both names.

realthunder, 2021-12-17:
Focus on the tree view, and press `Ctrl + F` to bring up the search box. Type in `Body004` to find the item. PS. if you have multiple files open, you can type in the full name `CMP_KaCh2_SFILT_6_4_tunable#Body004` for disambiguation.

wsteffe, 2021-12-17:
Ok thanks for hint.

Another situation in which I have some trouble is when, for some reason, a recomputation fails.
It seems to me that there is no indication of which is the broken Body/Feature.
In such cases it would be nice to have a graphical sign placed near the broken Body/Feature in the structure tree.

realthunder, 2021-12-17:
Normally the object with error will be marked in the tree. Can you please show an example of this?

wsteffe, 2021-12-17:
May be I have not seen the mark. 
At the moment I have not a broken design. I will look to it more carefully and eventually show an example when it happens again.

wsteffe, 2022-01-20:
Hello lupaz,  I forgot this issue but, after my last post I have observed that the error is marked in the tree as said by RT.
But the problem is that this mark, most of the times gets unnoticed (at least by me) because it is very small and nested inside the tree. You have to open all the bodies looking for this tiny mark.  It would be much  better if the body itself would be marked as failed and with a big mark (or being painted in red).
The messages in the report view  are also confusing. There is a lot of text related with the source code where the error was observed but this kind of info is of no interest for the end user. It could be useful to report a problem to the developers, but t should be hidden in the normal usage and made available only through a special setting. 
The important thing for the user is to identify the body/feature which failed so that he can adjust it. I am not so sure that this relevant info is reported in all cases but, when it happens, it is not clear because it is hidden inside a lot of other text lines and it does refer to the initial body name (in example Body001) instead of the user name. As suggested by RT the user may look for the name making a search (Ctrl + F) in the tree view. But why not making this search in the code before writing the error message ?