realthunder/FreeCAD #233: Part Design Hole - "Recompute failed!"

gkatev, 2021-12-19:
Hi, I'm trying to create a hole, but I'm getting a "Recompute failed! Please check report view." error (no extra info in the report view). As far as I can tell what I'm attempting to do is "legal" (?).

The document on which it happens: [latch-lock-wall.FCStd.zip](https://github.com/realthunder/FreeCAD/files/7740539/latch-lock-wall.FCStd.zip).

Select Sketch0003 and apply a hole, the error should appear. Furthermore, it actually does seem like Sketch003 is not necessary at all? If I just select the face that it lies on and apply a hole, it appears as it should -- I suppose it infers the hole's location from the face's circular feature? (pretty cool :-) )

Environment:
Arch Linux (installed via AUR package)
LinkDaily branch, commit #b527359 (latest at time of writing)

wohltat, 2021-12-19:
You have two circles in the Sketch003. Delete the additional one and it works.

By the way, the "Fillet feature" is an easier way to get round edges instead of cutting it with a "Pocket feature".

gkatev, 2021-12-19:
Oh indeed! Somehow there's an additional one with 0mm diamater... good catch -- my bad!

closed 2021-12-19 by gkatev