realthunder/FreeCAD #234: Theming question: Is it possible to set background of scroll bars in transparent dock?

xplwowi, 2021-12-20:
In my environment (KDE Plasma) scrollbars have always black background.
It is not related to the QT platform theme issue #168. I have no problems described there despite using Breeze style and HW acceleration.

I didn't find anything related in QSS files. I am able to freely edit all other parameters and the effects are visible on the screen. All available themes have ugly black scrollbars. There is no problem when the widget is highlighted and its transparency disappears. Then the background is correctly taken from the QScrollBar:*whatever* {background-color} properties.

[image]

xplwowi, 2022-01-14:
There seems to be a problem with my local builds.
For example - transparencies don't work at all for hovered over overlays. I compared my version to the one (appimage) that @realthunder recently uploaded to assembly3 releases. When using the same themes in the appimage version, the transparency works and scroll bars are styled correctly. I can't find a place where there can be a difference. Any old libraries on my system? Can anyone guide me to the solution? At the moment, the transparent overlays are rather unusable.

Environment:
Startup screen: FreeCAD 0.20, Libs: 0.20R30764 (Git)

OS: Ubuntu 21.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.30764 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.7
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Polish/Poland (pl_PL)

Transparency:
[image]

On-hover:
[image]

xplwowi, 2022-01-16:
And yet it turned out that I was wrong and using the Breeze theme is a problem...
Forced starting with a different theme makes the transparency work as it should.

This form of call worked:
QT_STYLE_OVERRIDE=oxygen ./FreeCAD

[image]

xplwowi, 2022-01-16:
Duplicate of #168

luzpaz, 2022-01-19:
@xplwowi this is not a dupe of 168

xplwowi, 2022-01-19:
IMHO it is. My environment is similar, the symptoms are the same, I tested use both Nvidia and Intel GPUs (no difference), GPU/software GL are both affected.
The only difference is that QT_QPA_PLATFORMTHEME=qt5ct freecad is not a solution. That's because qt5ct is also configured to use Breeze theme.
QT_STYLE_OVERRIDE=oxygen sets the theme natively/directly and it works well. I am not affected by the lack of native widgets appearance because I use the qss style that sets its own style anyway.