realthunder/FreeCAD #237: Recent commits (last couple days) -- old project with no warnings now reports many "skip invalid parent" warnings

scottmudge, 2021-12-21:
Perhaps you know of this, but just in case I wanted to report it.

One of my old projects which as of node b527359e936433d504b788ae54af12a60258781f had no warnings, now shows the below warnings when recomputing, after merging with the latest tip of LinkDaily:

```
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import001 skip invalid parent P_RHEG_NEMA#Part.Import001.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import skip invalid parent P_RHEG_NEMA#Part.Import.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import003 skip invalid parent P_RHEG_NEMA#Part.Import003.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import002 skip invalid parent P_RHEG_NEMA#Part.Import002.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import skip invalid parent P_RHEG_NEMA#Part.Import.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import001 skip invalid parent P_RHEG_NEMA#Part.Import001.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import002 skip invalid parent P_RHEG_NEMA#Part.Import002.
04:36:48  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import003 skip invalid parent P_RHEG_NEMA#Part.Import003.
04:36:49  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import010 skip invalid parent P_RHEG_NEMA#Part001.Import010.
04:36:49  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import009 skip invalid parent P_RHEG_NEMA#Part001.Import009.
04:36:50  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import020 skip invalid parent P_RHEG_NEMA#Part001.Import020.
04:36:50  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import009 skip invalid parent P_RHEG_NEMA#Part001.Import009.
04:36:50  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import010 skip invalid parent P_RHEG_NEMA#Part001.Import010.
04:36:50  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import020 skip invalid parent P_RHEG_NEMA#Part001.Import020.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import011 skip invalid parent P_RHEG_NEMA#Part002.Import011.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder002 skip invalid parent P_RHEG_NEMA#Part002.Binder002.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder001 skip invalid parent P_RHEG_NEMA#Part002.Binder001.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder skip invalid parent P_RHEG_NEMA#Part002.Binder.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import012 skip invalid parent P_RHEG_NEMA#Part002.Import012.
04:36:52  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import011 skip invalid parent P_RHEG_NEMA#Part002.Import011.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder skip invalid parent P_RHEG_NEMA#Part002.Binder.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import012 skip invalid parent P_RHEG_NEMA#Part002.Import012.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder001 skip invalid parent P_RHEG_NEMA#Part002.Binder001.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Binder002 skip invalid parent P_RHEG_NEMA#Part002.Binder002.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import013 skip invalid parent P_RHEG_NEMA#Part004.Import013.
04:36:53  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import008 skip invalid parent P_RHEG_NEMA#Part004.Import008.
04:36:54  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import008 skip invalid parent P_RHEG_NEMA#Part004.Import008.
04:36:54  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import013 skip invalid parent P_RHEG_NEMA#Part004.Import013.
04:36:55  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import016 skip invalid parent P_RHEG_NEMA#Part005.Import016.
04:36:55  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import005 skip invalid parent P_RHEG_NEMA#Part005.Import005.
04:36:55  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import017 skip invalid parent P_RHEG_NEMA#Part005.Import017.
04:36:56  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import006 skip invalid parent P_RHEG_NEMA#Part005.Import006.
04:36:56  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import018 skip invalid parent P_RHEG_NEMA#Part005.Import018.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import019 skip invalid parent P_RHEG_NEMA#Part005.Import019.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import007 skip invalid parent P_RHEG_NEMA#Part005.Import007.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import005 skip invalid parent P_RHEG_NEMA#Part005.Import005.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import006 skip invalid parent P_RHEG_NEMA#Part005.Import006.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import007 skip invalid parent P_RHEG_NEMA#Part005.Import007.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import016 skip invalid parent P_RHEG_NEMA#Part005.Import016.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import017 skip invalid parent P_RHEG_NEMA#Part005.Import017.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import018 skip invalid parent P_RHEG_NEMA#Part005.Import018.
04:36:57  <Part> SubShapeBinder.cpp(307): P_RHEG_NEMA#Import019 skip invalid parent P_RHEG_NEMA#Part005.Import019.
```

I have confirmed the project is exactly the same, nothing else has changed. Saving and reopening after recomputing with these warnings does not correct the warnings.

I'm assuming it is caused by this change in `SubShapeBinder.cpp`:

[image]

Is there a way to repair a file to prevent these warnings, assuming this is the intended current behavior?

scottmudge, 2021-12-21:
Actually, could this be caused by the additional suffixed `.` character in `resolvedSub`?

scottmudge, 2021-12-21:
I believe I've found a cause. It looks like all of my shapebinders/imports have an extra `.` character in the end of their `Support` or parent:

[image]

This appears to cause issues in `SubShapeBinder::update()`, where the `parentSub` string returned by `Context.getSubName(false)` also has this extra `.` character.

See the contents of the variable in debugging here:

[image]

Also in this screenshot you can see extra code I temporarily added to strip this extra suffixed `.` character away. When I do this, the warning/issue goes away, and it seems to perform normally.

scottmudge, 2021-12-21:
Further investigation shows the `.` characters are actually saved in the project itself, which may be the cause:

[image]

To fix it, I may add logic to strip away unnecessary `.` characters when parsing saved files. I'm not sure when the project was saved wrong, but there must have been a change to the export/save logic at one of my prior merges.

scottmudge, 2021-12-21:
So looking more deeply into this, the trailing period seems fine, and has existed all along.

However, when I add the following lines to `PropertyLinks.cpp` in the `Restore()` method, the warnings go away:

[image]

When I re-save the project, the `Import` objects/documents change like so:

[image]

This does not seem like the correct fix, or like a fix at all. It simply looks like it's simply aliasing the misinterpreted object with a new name, which `SubShapeBinder::update()` is able to handle appropriately. While it fixes the warnings, it doesn't seem like an ideal solution.

You know best about how all this fits together, but I just wanted to note what I am observing.

realthunder, 2021-12-21:
Can you please provide a file of yours to show the problem? I did fix a bug in the sub shape binder that cause it to not update when the parent placement changes. This is just a warning. It won't cause too much problem. Only that you may find the binder not update when its parent changes. Double click the binder will re-establish the context.

PS. The trailing dot is required if it is referring to an object. Without the dot, it indicates the last name is referring to a sub-element like `Edge` or `Face`. If it's `Face.` then this refers to an object named `Face`.

scottmudge, 2021-12-21:
Sure, I've attached a project file. 

[DishLid.zip](https://github.com/realthunder/FreeCAD/files/7756178/DishLid.zip)

Every single one of my projects I saved in the last week (or at least, all I have available) exhibit this warning. And every single "Import" or shape binder seems to have the same warning.

If it was just a one-off warning from a broken shape binder, then I'd not be concerned, but it's every single one.

commit dfcdba61 (2021-12-22): App: fix DocumentObject::resolveRelativeLink()

Fixes #237
@scottmudge
closed 2021-12-22 by realthunder