realthunder/FreeCAD #242: Compiling with error LNK2019: unresolved external symbol for boost regex using the libpack 12.5.4

apeltauer, 2022-01-02:
Hi @realthunder,
i started to try to compile your fork. But i get the following linker errors using the libpack 12.5.4

`1>Interpreter.obj : error LNK2019: unresolved external symbol "private: class boost::basic_regex<char,struct boost::regex_traits<char,class boost::cpp_regex_traits<char> > > & __cdecl boost::basic_regex<char,struct boost::regex_traits<char,class boost::cpp_regex_traits<char> > >::do_assign(char const *,char const *,unsigned int)" (?do_assign@?$basic_regex@DU?$regex_traits@DV?$cpp_regex_traits@D@boost@@@boost@@@boost@@AEAAAEAV12@PEBD0I@Z) referenced in function "int __cdecl getSWIGVersionFromModule(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?getSWIGVersionFromModule@@YAHAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
`

Something going on with the boost regex stuff.
Do you have any ideas how to solve this? Do i need a newer version of boost, or do i have to build boost with some special options?
Any hints are welcome....

Thanks...

StudioPetrikas, 2022-01-09:
Can confirm, getting ~50 of these regex errors, on Windows 11, building with VS2019

Output:
https://pastebin.com/HdZbF0bz

StudioPetrikas, 2022-03-14:
I no longer have this issue when building with FreeCADLibs_12.6.1_x64_VC17.7z