realthunder/FreeCAD #258: Transparent dock panels: last thing to improve (IMHO)

xplwowi, 2022-01-19:
Hi @realthunder
I love the change from last commits that removes frames and empty space around docked panels. Now everything looks seamless and almost perfect.
One last thing to have visual consistency is a narrow blank area at the right of the panels docked to the bottom of the screen.
The same panels dock to the very edge when the right side of the screen is used.
Take a look at the lower right corner - there is a gap between screen border and the report view.
[image]

xplwowi, 2022-01-19:
I was wrong again. Borders come directly from QT stylesheets:
"Breeze" is a disaster, "oxygen" has ugly borders, "fusion" and "plastique" are OK.

Oxygen - borders visible
[image]

Plastique, Fusion - seamless panels
[image]

dickshaydle, 2022-01-19:
I thought you meant the tab selection and status bar.
If there is only one tab and no status message then this area has a very low information / space  ratio.
Would be nice if the unused gray area of them would also be transparent, since especially vertical screen space is very valuable.

Also i have to agree it is really awesome how far the gui has come.

StudioPetrikas, 2022-01-21:
Can confirm about the empty rectangle in the bottom right corner. Widget inspector says it's OpenGL window (viewport).
No idea what causes that margin to appear.

xplwowi, 2022-01-24:
There is something wrong with OverlayBottom widget geometry.
[image]