realthunder/FreeCAD #260: segfault

wsteffe, 2022-01-27:
FreeCAD is crashing with the annexed document when I make a fillet on any edge of Pad001 (The current tip of DIEL_CAVITY_TOP). The crash occurs also when I try to change the depth of Pad001 or the depth of Pad.
These two depths were defined in a spreadsheet that has been removed. 
But I am not able clear this orphaned dependance without having a crash. 

[test.zip](https://github.com/realthunder/FreeCAD/files/7950102/test.zip)

xplwowi, 2022-01-27:
On KDE Plasma (Linux) I had no problem fixing the project by clearing orphaned formulas using Model->Data tabs. The workspace was empty due to dependency errors but there were no segfaults after recomputation.
1. Sketch->Data->Attachment Offset->Position->z: <\<Parameters>>.shield_thick / 2
2. Pad->Data->Length: <\<Parameters>>.b
3. Pad001->Data->Length: <\<Parameters>>.shield_thick / 2

Your dependency graph:
[image]

wsteffe, 2022-01-27:
I have just discovered that the problem was not related with the specific (uploaded) document but was introduced in last update of linkdaily that I have done this morning. In fact with this build FC was crashing with any document when making a pad.
Now I have reverted back my local build to a previous commit of January 16 (3b620b713e24c9637db8ea6ad1f1ee88ef13367a Sketcher: add action to orient sketch on the opposite side). Now the pad feature is working again.

wsteffe, 2022-01-27:
So RT a bug was introduced by a commit done between 16 January and today.

xplwowi, 2022-01-27:
I compiled the program with today's changes and  I can confirm that I have the same symptom when clossing a pad parameters dialog. Even with Cancel.
Problem is caused by today's commit "Gui: fix toggling show on top object" - cc57ae49b8ff6245cfd1a0fde43c90a1f54b13e2
Reverting it makes FreeCAD usable again.

commit ec0fcee4 (2022-01-27): Gui: fix crash on SubObjectT::normalize()

Fixes #260
@wsteffe
@xplwowi
closed 2022-01-27 by realthunder