realthunder/FreeCAD #263: [suggestion] to provide link to snap package

akshimassar, 2022-01-28:
I think your branches of FreeCad are used a lot. Personally I used freecad-linkstage3 aur package for arch linux, until I upgraded to python 3.10. I tried rebuilding it with previous python but failed since it requires a loot of libraries (including boost for python 3.9).

The point is I didn't know there is snap available, I even never heard of such package manager. I bet many others don't know too. And there are many people right now who cannot use even official freecad package.

Could you please replace standard README.md with some description of you fork (like it's done at https://snapcraft.io/freecad-realthunder) and provide a link to snap packages? It would greatly help those who want to use your branch and used to build it themselves.

akshimassar, 2022-02-10:
Could you please also point out that you have AppImage releases available in assembly workbench repo? It took for me a lot of time to figure that out, I guess it might take even longer for those who are not familiar with github and didn't get a hint from your upon existence of such images.

@realthunder what do you think?

luzpaz, 2022-03-20:
@realthunder should we mention snaps to the https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11 page perhaps?

commit 64efac6b (2022-03-22): Change ReadMe

Closes #263
closed 2022-03-22 by realthunder