realthunder/FreeCAD #264: Bug: Visibility controls no longer functioning

jj-uk, 2022-01-29:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

None of the visibility controls are working.

I can open and save the file in 0.19 (official), then open it in LinkStage3, but after a few minutes of using the application, the visibility tools stop working again.

What can I do to repair the file? 

[cp-linkBranch.20220129-170948 - Copy.zip](https://github.com/realthunder/FreeCAD/files/7964937/cp-linkBranch.20220129-170948.-.Copy.zip)

DerTFL, 2022-02-01:
Hi! Can you try method described in this issue comment?: https://github.com/realthunder/FreeCAD_assembly3/issues/1087#issuecomment-1022243000

jj-uk, 2022-02-17:
Yes, that works for me, thanks,

closed 2022-02-17 by jj-uk