realthunder/FreeCAD #270: [Bug] cannot set spreadsheet cell value to 0.07"

not-a-number-dot-org, 2022-02-08:
New file
New spreadsheet
Enter 0.07" in any cell
Select cell
Content has been set to : 0.07000000000000001"

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)

dickshaydle, 2022-02-08:
That's kind of normal. To be precise every spreadsheet software is doing that because it is rooted in the way decimal numbers are represented. In the computer we use binary numbers in a floating point representation. 
Maybe you know that 1/3 is 0.33333... which is not representable with a finite number of digits in decimal. It's kind of similar that 0.7 is not representable with a finite number of digits in binary. Since we don't have infinit digits there will be an error / difference to the "real" 0.7. If you then convert it back from 0.7 in binary to 0.7 decimal you see the error. 
That's exactly the difference you have reported.

Here is a similar case explained:
https://stackoverflow.com/questions/34081470/why-is-0-29999999999999998-converted-to-0-3

not-a-number-dot-org, 2022-02-08:
I know.
But if I enter =1/3 in a cell and inspect it, its content remains 1/3, it is not replaced by 0,3333...
Shouldn't the cell remain to 0.07 if I enter =0.07 ?

If I enter 0.07 well... it's fine to replace it by its internal value.
But if I enter =0.07, shouldn't it remain 0.07 ?

Shouldn't =0.07 be treated as an expression and 0.07 as a value ?

btw '=' was missing in my 1rst post

closed 2023-10-29 by not-a-number-dot-org