realthunder/FreeCAD #273: Fillet leads to weird geometry error

akshimassar, 2022-02-08:
[image]
[heater_handle.FCStd.zip](https://github.com/realthunder/FreeCAD/files/8026657/heater_handle.FCStd.zip)
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.20.30786 (Git)
Build type: Release
Branch: makepkg
Hash: db6211fa22dc5145d69355a3aecf143e5f1b5f4b
Python version: 3.10.1
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

akshimassar, 2022-02-09:
I'm so sorry, I messed up command names and run freecad from arch repo instead of one from snap. Closing the issue since it is not related to your repo.

akshimassar, 2022-02-09:
Well, I guess it was too fast to close it. I've got the same problem with branched version.
[image]

OS: Ubuntu Core 20
Word size of FreeCAD: 64-bit
Version: 2022.209.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.02.09-edge)
Hash: 5f7d60b65639efb574b6a8a5238856f11f36910f
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

akshimassar, 2022-02-09:
Should I report that to upstream?

akshimassar, 2022-02-09:
Chamfering certain edge (selected at the image) even leads to crash right away.
[image]
```
<00:48:43> 22 akshi-desktop ~
$ freecad-realthunder
FreeCAD 0.20, Libs: 2022.209R14555 (Git shallow)
Adding snap-specific PYTHONPATH to sys.path: /home/akshi/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/7/lib/python3.8/site-packages:/snap/freecad-realthunder/7/usr/lib/python3/dist-packages
4.319554 <asm3.main> init_gui.py(14): no solver backend found
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f582dbde210]
#1  0x7f5826c7eddb in TopLoc_Location::Multiplied(TopLoc_Location const&) const from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKMath.so.7+0x1b
#2  0x7f5825900f73 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKBRep.so.7+0x33
#3  0x7f58259182e6 in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKBRep.so.7+0x96
#4  0x7f582735f69f in ChFi3d_Builder::IntersectMoreCorner(int) from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKFillet.so.7+0x89f
#5  0x7f582736b53d in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKFillet.so.7+0x883d
#6  0x7f582730e880 in ChFi3d_Builder::PerformFilletOnVertex(int) from /snap/freecad-realthunder/7/usr/lib/x86_64-linux-gnu/libTKFillet.so.7+0x3e0
```

realthunder, 2022-02-09:
There is a known fillet/chamfer problem in OCC 7.5. FreeCAD offical prebuilt images uses a patched version that temporarily fixed the problem, and so did my AppImage release. Can you please try these and see if it fixed your problem?

akshimassar, 2022-02-10:
Sorry, but where can I get your AppImage release? I would be happy to use it.

akshimassar, 2022-02-10:
I took `FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64.AppImage` and can confirm it crashes upon chamfering designated edge.

akshimassar, 2022-02-10:
[image]
I also got some artifact while trying to fillet the edges.
[heater_handle_weird_fillet.FCStd.zip](https://github.com/realthunder/FreeCAD/files/8040245/heater_handle_weird_fillet.FCStd.zip)

realthunder, 2022-02-10:
I can confirm the problem still exists with OCC 7.6, both the ugly fillet and segfault when doing a chamfer.

akshimassar, 2022-02-12:
Any chance to fix that? Should I report the issue to upstream?

wsteffe, 2022-02-24:
I may confirm that chamfering that edge cause a crash with LinkDaily.
Furthermore fillets are weird as shown in the following image.
[image]
 
I have fixed that in my branch https://github.com/wsteffe/FreeCAD/tree/fixBooleansAndFillets

The fixed geometry is shown in the following image:

[image]

The fix is quite simple: I have applied  ShapeUpgrade_ShapeDivideClosedthe to the Pad feature (and also to the Revolution) as done with Ellipsoid Primitive in my PR 283.

But there is a price to pay for Pad and Revolution because these features are used a lot in many project.
Most structures will probably have to be fixed if this patch is merged because many geometries will be modified by the splitting of closed surfaces and many references will be lost. TN (at the moment) is not able to recover it and it must be fixed manually.
In fact I had to redefine all the feature in the document heater_handle_weird_fillet.FCStd before applying the fillet.