realthunder/FreeCAD #274: [GUI] Pie menu icons broken/invalid

scottmudge, 2022-02-11:
### Build Affected
Current tip of LinkDaily branch (91c5e00f6fd7c994bb7e73c188f6bad57a4928eb)

### Description

The pie menu icons are showing up incorrectly in the latest tip. Minor issue. Icons look fine elsewhere.

[image]

Just thought I would make note of it.

realthunder, 2022-02-14:
I can't duplicate the error. Can you paste your about information here?

scottmudge, 2022-02-14:
I believe the problem may be constrained to Windows. See below as to why. I compiled off the node in the original issue report, as I normally do. I've not made any changes to the code, and the problem seemed to appear in one of the more recent commits (maybe in the last week or so).

I'm also noticing a new warning in the report view which appears to show invalid directory path parsing:

`05:17:46  BrowserView: file does not exist: C:/AssemblyWorkbenchBrowserView: file does not exist: C:/AssemblyWorkbench`

Perhaps it's an issue with Windows directory/paths using backslashes, where double backslashes are missing somewhere.

scottmudge, 2022-02-15:
Alright starting to debug this now.

#### Point 1

That warning I posted in my previous comment is actually a bug in the asm3 module (using the latest tip from your repo). It looks like it's associated with commit 4d815377e1e33dddd21e6250ffe4ea589f27c027

#### Point 2

The issue with the icons appears to be that (inside the PieMenu's init() callstack) the icon path/name is being set to a memory address (the actual text is a memory address) rather than a filename:

[image]

I'll debug this further and let you know.

scottmudge, 2022-02-15:
Actually, perhaps that's normal for it to be a memory address? It's odd it seems to be treated it is a filename in this function, but perhaps the icon is meant to be already loaded in the cache and associated with the memory address as the key?

Regardless, the "filename"/memory address is being assigned here:

[image]

commit 194af51b (2022-02-17): Gui: fixes workbench command icon problem

Fixes #274
@scottmudge
closed 2022-02-17 by realthunder