realthunder/FreeCAD #275: Issue with FEM example

alowell-ssl, 2022-02-12:
I'm trying to run the "Electrostatics Capacitance Two Balls" example from the FEM workbench, and it doesn't work properly.  I can successfully write the config files for the Elmer simulation, but when I click "run" I get a popup window that says "Error: Result file not found."  

The report view says:
15:40:32  Solver binary path (returned from binary getter): /snap/freecad-realthunder/7/usr/bin/ElmerSolver 
15:40:32  Exception in thread Thread-379:
15:40:32  Traceback (most recent call last):
15:40:32    File "/snap/freecad-realthunder/7/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
15:40:32      self.run()
15:40:32    File "/snap/freecad-realthunder/7/usr/lib/python3.8/threading.py", line 870, in run
15:40:32      self._target(*self._args, **self._kwargs)
15:40:32    File "/snap/freecad-realthunder/7/usr/Mod/Fem/femsolver/task.py", line 118, in protector
15:40:32      self.run()
15:40:32    File "/snap/freecad-realthunder/7/usr/Mod/Fem/femsolver/elmer/tasks.py", line 163, in run
15:40:32      self.solver.ElmerResult.read(postPath)
15:40:32  TypeError: argument 1 must be str, bytes or bytearray, not None

I looked in the directory where the simulation config files are written.  I see that no result file is generated.  If I manually run the ElmerSolver binary (using the binary that is packaged with the freecad-realthunder snap) in this directory, it generates a result file (case_t0001.vtu) which I can open in paraview where it is clear that the simulation worked.

So it seems that FreeCAD is not successfully running ElmerSolver.

ppd, 2022-02-15:
It's because ElmerSolver fails to start because orted fails to start for some reason:

```console
maxi@maxi-desktop:/tmp/femqzupq9g1$ ElmerSolver
ELMER SOLVER (v 9.0) STARTED AT: 2022/02/15 13:46:44
--------------------------------------------------------------------------
Sorry!  You were supposed to get help about:
    no-plugins
But I couldn't open the help file:
    /usr/share/pmix/help-pmix-runtime.txt: No such file or directory.  Sorry!
--------------------------------------------------------------------------
[maxi-desktop:204011] [[INVALID],INVALID] ORTE_ERROR_LOG: Unable to start a daemon on the local node in file ess_singleton_module.c at line 716
[maxi-desktop:204011] [[INVALID],INVALID] ORTE_ERROR_LOG: Unable to start a daemon on the local node in file ess_singleton_module.c at line 172
--------------------------------------------------------------------------
It looks like orte_init failed for some reason; your parallel process is
likely to abort.  There are many reasons that a parallel process can
fail during orte_init; some of which are due to configuration or
environment problems.  This failure appears to be an internal failure;
here's some additional information (which may only be relevant to an
Open MPI developer):

  orte_ess_init failed
  --> Returned value Unable to start a daemon on the local node (-127) instead of ORTE_SUCCESS
--------------------------------------------------------------------------
--------------------------------------------------------------------------
It looks like MPI_INIT failed for some reason; your parallel process is
likely to abort.  There are many reasons that a parallel process can
fail during MPI_INIT; some of which are due to configuration or environment
problems.  This failure appears to be an internal failure; here's some
additional information (which may only be relevant to an Open MPI
developer):

  ompi_mpi_init: ompi_rte_init failed
  --> Returned "Unable to start a daemon on the local node" (-127) instead of "Success" (0)
--------------------------------------------------------------------------
*** An error occurred in MPI_Init
*** on a NULL communicator
*** MPI_ERRORS_ARE_FATAL (processes in this communicator will now abort,
***    and potentially your MPI job)
[maxi-desktop:204011] Local abort before MPI_INIT completed completed successfully, but am not able to aggregate error messages, and not able to guarantee that all other processes were killed!
maxi@maxi-desktop:/tmp/femqzupq9g1$ ElmerSolver
ELMER SOLVER (v 9.0) STARTED AT: 2022/02/15 13:47:32
--------------------------------------------------------------------------
Sorry!  You were supposed to get help about:
    no-plugins
But I couldn't open the help file:
    /usr/share/pmix/help-pmix-runtime.txt: No such file or directory.  Sorry!
--------------------------------------------------------------------------
[maxi-desktop:209606] [[INVALID],INVALID] ORTE_ERROR_LOG: Unable to start a daemon on the local node in file ess_singleton_module.c at line 716
[maxi-desktop:209606] [[INVALID],INVALID] ORTE_ERROR_LOG: Unable to start a daemon on the local node in file ess_singleton_module.c at line 172
--------------------------------------------------------------------------
It looks like orte_init failed for some reason; your parallel process is
likely to abort.  There are many reasons that a parallel process can
fail during orte_init; some of which are due to configuration or
environment problems.  This failure appears to be an internal failure;
here's some additional information (which may only be relevant to an
Open MPI developer):

  orte_ess_init failed
  --> Returned value Unable to start a daemon on the local node (-127) instead of ORTE_SUCCESS
--------------------------------------------------------------------------
--------------------------------------------------------------------------
It looks like MPI_INIT failed for some reason; your parallel process is
likely to abort.  There are many reasons that a parallel process can
fail during MPI_INIT; some of which are due to configuration or environment
problems.  This failure appears to be an internal failure; here's some
additional information (which may only be relevant to an Open MPI
developer):

  ompi_mpi_init: ompi_rte_init failed
  --> Returned "Unable to start a daemon on the local node" (-127) instead of "Success" (0)
--------------------------------------------------------------------------
*** An error occurred in MPI_Init
*** on a NULL communicator
*** MPI_ERRORS_ARE_FATAL (processes in this communicator will now abort,
***    and potentially your MPI job)
[maxi-desktop:209606] Local abort before MPI_INIT completed completed successfully, but am not able to aggregate error messages, and not able to guarantee that all other processes were killed!
```

OpenMPI fails to start, so much is certain. Probably because Apparmor denies something it needs to start. Maybe this can be fixed by adding some interface.

ppd, 2022-02-15:
Looks like it does not find the pmix libraries. A symlink should help. Let me try this.

closed 2022-02-16 by realthunder