realthunder/FreeCAD #276: Compile errors due to include order with latest changes, on Windows

scottmudge, 2022-02-15:
The include/import order is incorrect in `DocumentObject.cpp`, leading to compiler errors on Windows.

Because `PreCompiled.h` imports `FCConfig.h`, which contains the DLL import/export macros, if this is included *after* other headers which use these import/export macros, compiler errors occur.

This is limited to Windows.

The fix is to re-order the imports such that `PreCompiled.h` or `FCConfig.h` is included before any other headers:

[image]

realthunder, 2022-02-15:
Wow, what a silly mistake I was making. I don't understand how it got there. I don't remember doing that. Must have triggered some auto fix feature in the intellisense somehow. Thanks for the heads up!

closed 2022-02-15 by realthunder

scottmudge, 2022-02-15:
It happens. If I use the feature which automatically adds missing imports/includes with Visual Assist, it seems to put them randomly at the top of the file, sometimes before other includes. Probably just an automatic bug.

scottmudge, 2022-02-15:
Thanks for the quick fix! 👍

commit 0033a79a (2022-02-15): Fix build error

Fixes #276