realthunder/FreeCAD #278: Broken boolean when applied to a WB Part Primitive (ellipsoid)

wsteffe, 2022-02-17:
Hello RT, may you please help me.
I have a problem with the annexed document which is not properly exported into a step file.
The problem is with the boolean (the last item in the Body named DIEL_CAVITY).
Initially this structure was exported well but it got broken after I changed a little bit the definition of DatumPlane which is used as a support of the first Sketch in DIEL_CAVITY.
It is not the first time it gets broken in this way but other times I was able to fix it making again the subshape binder to the ellipsoid and the subsequent boolean. This trick doesn't work any more. I have recreated these objects several times but the Boolean can not be fixed.

[issue.zip](https://github.com/realthunder/FreeCAD/files/8090850/issue.zip)

wsteffe, 2022-02-18:
When I try to export the Part (named CMP_ballTriplet) into a step file FC produce following messages in report view:

07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face14
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face15
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face16
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face17
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face18
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face19
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face2
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face20
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face21
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face22
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face23
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face3
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face4
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face5
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face6
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face7
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face8
07:02:42  <Import> ImportOCAF2.cpp(1211): Failed to get subshape Face9

wsteffe, 2022-02-18:
I have just recreated the same structure from scratch in a new FC document and the step export is working again. So I must conclude that FC document structure gets corrupted after reediting or changing a little bit a small part of it.  It is very bad behavior because one has to restart a design from the beginning again and again  when such kind of problem occurs. This was a very simple document but the problem could be very serious if it affects a complex project,

wsteffe, 2022-02-18:
Another (slightly minor) issue which I observed in the uploaded document is the resetting to 1 of the Number of Occurences in the Polar Pattern each time this feature is reopened. In this structure I have set Number of Occurences=2  (=1 doesn't make anything). The automatic reset of the Number of Occurences changes (and actually brakes) the document structure.

wsteffe, 2022-02-18:
Hello RT,  I have used the new (recreated) FC document in an electromagnetic optimization. Each optimization step changes a little bit the parameters stored in the Spreadsheet named "Variables". In this process the boolean has been broken again without any apparent reason. Here annexed you will find a file named "ballTriplet_good" containing the structure as it was just before it was broken and a file named "ballTriplet_bad"  with the broken structure.

If you export in a step file the Part (named CMP_ballTriplet)  stored in "ballTriplet_bad" and reimport it again you fill find that the boolean is wrong. 

[issue2.zip](https://github.com/realthunder/FreeCAD/files/8096237/issue2.zip)

wsteffe, 2022-02-18:
I have just realized that Ellipsoid primitive exists also in PD WB (while I was using that one from Part WB).
So I made another version (here annexed) of the same structure (with small changes) using PD Ellipsoid. 
But problem with boolean (ellipsoid part not being exported in step file) persists.
[issue3.zip](https://github.com/realthunder/FreeCAD/files/8097335/issue3.zip)

realthunder, 2022-02-18:
The warning messages when exporting STEP are not related to the wrong export shape. I'll fix that later. If I select the `Boolean` and export, I got this when import it back,

[image]

If you use `Part -> Check geometry` on the `Boolean` shape, it will report the shape being invalid. Probably an OCC problem. Based on the report, it seems to be related to the seam edge on the ellipsoid. If I rotate it 90 degrees (either the elliposid itself, or any of the binder in the path), the export becomes good, although the geometry checker still reports error (but without the seam edge ones).

[image]

wsteffe, 2022-02-19:
Hello RT,
 "The warning messages when exporting STEP are not related to the wrong export shape."
But I have observed that when those messages are given the exported boolean is broken (missing ellipsoid contribute) while when they are not the the exported boolean is good.

But now I have new info because something very strange happened today. I have just reopened the two uploaded files "ballTriplet_good.FCStd" and "ballTriplet_bad.FCStd" and now with both the exported boolean is good. I have the following messages in the bash terminal in which I entered the FreeCSAD command:

> 1.43e-07 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Face26 not found
> 4.58203 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Edge30 not found
> 8.00596 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Face26 not found
> 9.97387 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex28 not found
> 11.4859 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex37 not found
> 13.3021 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex38 not found
> 14.4861 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Edge58 not found
> 16.2702 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex39 not found
> 21.43 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex33 not found
> 23.6461 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex35 not found
> 24.558 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Vertex40 not found
> 25.5738 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Edge52 not found
> 26.55 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Edge56 not found
> 29.278 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Face10 not found
> 423.519 Selection.cpp(1903): Sub-object ballTriplet_PDprimitive#Part.Edge21 not found
> 428.236 Selection.cpp(1903): Sub-object ballTriplet_PDprimitive#Part.Edge8 not found
> 512.366 Selection.cpp(1903): Sub-object ballTriplet_bad#Part.Face25 not found

But I do not know in which moment they were generated.
Is it possible that FC has, in some way, self corrected a wrong data structure ?

I have also made a geometry check for both files selecting the body DIELECTRIC_CAVITY (which contains the boolean). The result is more or less the same in both case: an invalid edge and an invalid face.
But please notice that face and edge have changed: Edge6 became Edge30, Face1 became Face10.
The transition from the good document to the bad one was determined by a small changes of dimensions in "Variables" spreadsheet. It was not the first change but it was that one which caused the failure. I suspect that it has to do with a topological change in the structure that occurred for this specific perturbation. I think that the OCC structure was still valid (the Geometry Check always reports a couple of errors probably due to a strange structure) and that the export problem was generated by the topo-naming (which now should have corrected the names). I thing that the warning messages was because the exporter was looking for the geometrical data using wrong names. 

[image]
[image]

realthunder, 2022-02-19:
The above commit shall fix the warning message. It is only triggered for colorful body that has manual specified child export. It is not a fatal error, and shouldn't have any effect on the export shape.

wsteffe, 2022-02-19:
With a more careful check I have seen that the geometry of boolean in the "ballTriplet_Bad" is really broken (not just exported bad). In the next picture you may see that the seam edge is dangling on one side.
It should be connected to the circular green edge, which should have been divided in two (but it hasn't)

[image]
.

wsteffe, 2022-02-19:
But a strange thing is that the structure in "ballTriplet_Bad" has been fixed (I do not know how) after I played a little bit inside of it. I have made a few geometry checks, I reopened and closed (without any change) a couple of features. Here annexed the fixed geometry. I am not able to reproduce what I have done to get it. [fixed.zip](https://github.com/realthunder/FreeCAD/files/8101819/fixed.zip).
I am still suspecting that the difference is in what is displayed and not in the real result of the boolean operation. I haven't changed anything and I can not believe that the boolean has been done (in a second time) in the correct way.

wsteffe, 2022-02-19:
I am sorry. I made a wrong comparison. The starting point of last uploaded document (ballTriplet_bad_fixed), was not the same  ballTriplet_bad uploaded before. It was another version in which I had replaced Ellipsoid from Part WB with Ellipsoid from PD.

wsteffe, 2022-02-19:
Now I am thinking that the root of all this problem is a bed definition of Ellipsoid geometry.
If you check you may see that a simple Ellipsoid as it is created using (in example) PD primitive doesn't pass the geometry check.
This surface is defined using only one seam Edge. The single seam edge can be used to build a degenerated closed wire running along it in both direction. But I think that is is a very bad (degenerate) representation of the ellipsoid boundary. It would be much better (more robust) if the surface were split in two halves each one of them being represented with two orthogonal meridians (a seam edge and a boundary circle). If you tell me which is the part of the code that generates the ellipsoid, I would like to add to it a few lines to do that (using ShapeUpgrade_ShapeDivideClosed as explained at https://dev.opencascade.org/doc/overview/html/occt_user_guides__shape_healing.html#occt_shg_4_2_1).

[image]
.

wsteffe, 2022-02-19:
At this point I think it is better to close this issue and to open a new addressing the invalid geometry generated by ellipsoid primitive.

wsteffe, 2022-02-23:
This issue will be closed after merging of PR 283.

closed 2022-02-24 by wsteffe