realthunder/FreeCAD #280: Pocket not working

not-a-number-dot-org, 2022-02-19:
Pocket not working above 92mm

80mm:
[image]
Full width pocket :
[image]
16:05:32  62798.9 <App> Document.cpp(4231): Failed to recompute bug#Pocket: Pocket: Resulting shape is not a solid
16:05:32  Recompute failed! Please check report view.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git)
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)

[bug.zip](https://github.com/realthunder/FreeCAD/files/8102399/bug.zip)

wohltat, 2022-02-19:
--------------------
```
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

wohltat, 2022-02-19:
I cannot open the file. When i try i get this error:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: basic_ios::clear: iostream error
```

----------------------------------------
``````

OS: Linux Mint 20.1 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
``````

not-a-number-dot-org, 2022-02-19:
oups... missclick

See [#1099](https://github.com/realthunder/FreeCAD_assembly3/issues/1099#issue-1121489031) on opening on Linux files created on Windows

wsteffe, 2022-02-24:
bug.zip file is bad. Can not be extracted.

not-a-number-dot-org, 2022-02-24:
[bug.FCStd.gz](https://github.com/realthunder/FreeCAD/files/8132555/bug.FCStd.gz)

wsteffe, 2022-02-24:
Same problem as wohltat:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: basic_ios::clear: iostream error

wohltat, 2022-02-24:
I can reproduce the bug now with the gzip file.

The strange thing is that the at first i could not create a pocket wider than 91.4mm.
But then after successively increasing the pocket length by 0.01mm im now over 93mm.
In other words, there seems to be no fixed length that breaks the pocket. It seems that the limit can be pushed further indefinitely.

wohltat, 2022-02-24:
If you move that yellow edge of the `Sketch002` outside of the body then it works.
[image]
So in a way this is just a workaround.
 
I guess the behaviour reported in this issue is still a bug since it should work anyways.

not-a-number-dot-org, 2022-02-24:
I made it some other way.
I just posted to report the bug.

[image]

realthunder, 2022-03-10:
I think this is an OCC bug related to cutting with overlapping seam edge highlighted below. The overlapping happen between the mirror shape and the pocketing tool shape. You can use an expression to add some tolerance to work around the problem as shown below. The syntax for referencing the mirror edge is a bit complex. You can simply type `Mirrored001.Edge38 + Tab` and auto completer will fill in the rest.

[image]

[image]