realthunder/FreeCAD #281: Treeview reordering not working for Linked Objects

barry30, 2022-02-21:
Hi,
It seems if you have several Linked Objects, you can't reorder one of them with respect to other Linked Objects.  They can be moved above/below Bodies, but not other Linked Objects.

OS: Devuan GNU/Linux 4 (chimaera) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

realthunder, 2022-02-22:
Can you please provide a file showing the problem?

barry30, 2022-02-22:
In Part2, I can't move Link(Body4) to be above Link(Body3) but in Part1, I can move Body2 to be above Body1.

I believe this behavior is the same, where using a Part or not.

(File renamed to .zip for upload)
[tree_reorder_test.zip](https://github.com/realthunder/FreeCAD/files/8116415/tree_reorder_test.zip)

barry30, 2022-03-05:
Did you want something else other than a test file, or maybe you haven't had time to look at it?

realthunder, 2022-03-06:
Yeah, a bit busy. I'll definitely check this out.

realthunder, 2022-03-06:
I can't seem to reproduce the problem. Can you please make a screen cast? Also, please send me you user config file, `Tools -> Edit parameters -> Export`.

barry30, 2022-03-06:
Please find attached config file and video.

[freecad_link_daily_params.txt](https://github.com/realthunder/FreeCAD/files/8192858/freecad_link_daily_params.txt)

https://user-images.githubusercontent.com/100158063/156929172-985b06eb-4cbb-41fd-857b-a4bc24c5cb52.mp4

realthunder, 2022-03-07:
Still can't reproduce the problem. Are you for some reason holding the `Alt` key while dragging?

closed 2022-03-07 by barry30