realthunder/FreeCAD #282: Cannot find source file:      mtextedit.h

J-Dunn, 2022-02-21:
I seem to be hitting this rather old build bug: 

https://github.com/FreeCAD/FreeCAD/commit/968faa511a

should the trivial fix be added here?

realthunder, 2022-02-22:
It seem the referred commit has already been merged. Are you compiling LinkDaily or LinkStage3 branch?

J-Dunn, 2022-02-22:
Thanks.  I initially did a cmake with an incorrect -S option ( it pointed to a FreeCAD/FreeCAD tree ). I then did make with the correctly source specified thunder tree and got this error.  Since this bug was removed years ago I don't quite see where this came from, but deleting the entire thunder tree and gitting a new clone cleared the problem.

Maybe there is cmake command I could have run clear the CMake files instead of just deleting. 
Anyway sorry for the noise, it does not seem to be bug in your work.

closed 2022-02-22 by J-Dunn