realthunder/FreeCAD #291: Keybindings being weird and some not working at all after upgrade to LinkStage daily 2022.111

VorpalBlade, 2022-03-12:
After upgrading to the newest LinkStage daily keyboard shortcuts (at least in the sketcher) have changed. Several of them doesn't work at all:

* Pressing = to bring up the expression editor when editing the value of a constraint doesn't work, it inserts a literal = instead.
* Several of the shortcuts have been rebound to hard to reach keys. For example Vertical length constraint used to be `Shift-V` which is easy to reach one-handed with the other hand on the mouse. Now it is `I` which requires moving the hand to the other side of the keyboard, or moving the right hand to the keyboard. `Shift-D` for length constraint changing to `K,D` is even worse.

Version info:
```
OS: Arch Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

wohltat, 2022-03-12:
I noticed the same and also wonder why the keybindings have changed. I also find them worse than before.
The `i` and `l` keys are very un-intuitive. And the two key shortcuts (like K,D) really break the workflow in my opinion.
It even broke the polygon key in the sketcher. It is neither on the key as before nor on the new displayed keybinding.

I think that if such drastic changes are made to the keybindings then this should go with a selectable set of keybindings similar to one can choose different mouse navigation styles. So those who don't like the new set of keys can easily switch back.

realthunder, 2022-03-18:
Those changes are done upstream. You can change it back using `Customize -> Keyboard`. Maybe one of you can make a `Preset` for the old keybindings and share with others?

wohltat, 2022-03-20:
I would like to make a preset but i'm not sure how to do it.
Do you mean change the keybindings in `Customize -> Keyboard` and then use the `Parameter Editor` to export Settings?

At the moment there are only very few shortcuts visible in the Parameter Editor.

I guess if i only want to export keybindings and not other settings i have to manually edit the exported settings file, correct?

realthunder, 2022-03-21:
If you are using Linux, please try [snap](https://snapcraft.io/freecad-realthunder) build. The keyboard settings if latest version offers more efficient settings.

https://user-images.githubusercontent.com/1207888/159208388-f9e7d4b6-b361-46ac-9fa3-1e72264a2dfa.mp4

VorpalBlade, 2022-04-06:
@realthunder Do you have a flatpak instead of a snap? Snap is not available in the default repositories on Arch (I would have to build and install it from AUR). From what I understand pretty much only Ubuntu and some distros derived from it uses snap.

VorpalBlade, 2022-04-11:
While that is a great idea on paper there are two issues with it:
1. Both freecad-linkstage3-bin and freecad-linkstage3-git conflict with standard freecad. I still have the normal release installed and would prefer being able to install alongside that.
2. I was hoping to avoid the long compilation time of a large project like FreeCAD (which the `-bin` version of the package would, but it appears to be built by the PKGBUILD author, rather than using an upstream binary build, not sure I trust that).

There is also a freecad-assembly3-git in the [search results](https://aur.archlinux.org/packages?O=0&K=freecad), but it appears to be very outdated.

luzpaz, 2022-04-11:
Can we close this issue since it's established that the change came from upstream?