realthunder/FreeCAD #293: Building FreeCAD Link for Windows

StudioPetrikas, 2022-03-15:
Hi, I'm your local idiot and I've lost the ability to build FreeCAD Link from source.

The build process goes through correctly without error. Launches and works correctly. However, the splash screen is still Vanilla FreeCAD and the setup uses `user.cfg` instead of `link.user.cfg`. 

I'm basically ending up with FreeCAD, with FreeCAD Link features.
Tried both 
FreeCADLibs_12.5.5_x64_VC17.7z 
and
FreeCADLibs_12.6.1_x64_VC17.7z 

Clearly I'm missing something, but I can't figure out what. Thanks in advance, and sorry for opening an issue ticket.

realthunder, 2022-03-15:
Those customization are done through the FreeCAD branding facility. My daily release configuration can be found [here](https://github.com/realthunder/FreeCADMakeImage/tree/master/conda/branding/asm3-daily/branding). To install it, copy `branding.xml` to the `bin` directory, and copy the rest to the root directory of your FreeCAD build.

BTW, you need to manually replace all `_FC_XXX` entry in the `branding.xml` for a sensible version and build date. And of course, you can make other customization, and supply your own image files.

StudioPetrikas, 2022-03-15:
Ah, makes perfect sense now. Thank you very much for the answer.
Works perfectly, loads up the correct cfg.

closed 2022-03-15 by StudioPetrikas