realthunder/FreeCAD #294: Disable multi-Key sequence delay for shortcuts with no contextually relevant multi-key possibilities

scottmudge, 2022-03-17:
I noticed a (to me) very unpleasant delay when using keyboard shortcuts in the sketcher after merging with the daily branch recently.

I use the "Radius" ("R") shortcut quite frequently, and initially it felt like the application was now lagging. I'm fairly sure there was 0 delay before this merge for the radius shortcut, but now it is applied to all shortcuts.

Would it be possible to skip the timer/delay for keys or shortcuts which have only a single possible outcome given the context in which they are used? For instance, in the sketcher there are many default keys for constraints which only have a single binding, without any sequence assigned to the base keyboard shortcut.

realthunder, 2022-03-17:
The delay is configurable through `Customize -> Keyboard -> Multi-key sequence delay`. The delay is only used if there is ambiguity, which includes any key sequence with the same prefix. You can check it with the command complete as shown below, typing the shortcut sequence as keyword.

[image]

scottmudge, 2022-03-17:
Yes I know I've already set it to zero.

My point was to set it to zero by default for **only** keys which have no possible secondary multi-key sequence in a given context/workbench.

Thus eliminating the delay for most of the shortcuts which now feel sluggish, while **also** allowing for multi-key shortcuts to be used.

scottmudge, 2022-03-17:
I see, I assumed the delay was mostly for programmatic reasons, but I see it is mostly for providing clarity to the user about the active item.

I'll leave it set to 0.

closed 2022-03-17 by scottmudge