realthunder/FreeCAD #297: building linkdaily on macos `make install` issue

ipatch, 2022-03-23:
hello, 👋

i'm one of the maintainers of the homebrew-freecad tap, and am attempting to build your _linkdaily_ branch on macos and am running into some issues related to _pybind11_

i'm not experiencing any issues with the `make` command, but rather the `make install`.

i believe the issue stems from the `CMakeLists.txt` within the `/src/MacAppBundle` directory, and I do not experience these issues compiling and installing the upstream freecad, ie. github.com/freecad/freecad master branch

---

i'm setting up my build directory with the below cmake command

```
cmake \
-DHOMEBREW_PREFIX="$bp" \
-DPYTHON_EXECUTABLE="$bp/bin/python3" \
-DPYTHON_INCLUDE_DIR="$bp/opt/python@3.9/Frameworks/Python.framework/Headers" \
-DBUILD_QT5=1 \
-DCMAKE_PREFIX_PATH="$pthxercesc;$pthocc;$pthmedfile;$ptheigen;$pthqt;$pthcoin;$pthtbb;$pthspnav;$pthnetgen;$pthgraphviz" \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DBUILD_ENABLE_CXX_STD=C++17 \
-DFREECAD_CREATE_MAC_APP=1 \
-DFREECAD_USE_PYBIND11=1 \
-DBUILD_FEM_NETGEN=1 \
-DCMAKE_INSTALL_PREFIX="$code/freecad-git/installs/mar22.01.22.linkdaily" \
../../freecad-src
```

using that above cmake command sets up the below logic in the `cmake_install.cmake` file

```
if("x${CMAKE_INSTALL_COMPONENT}x" STREQUAL "xUnspecifiedx" OR NOT CMAKE_INSTALL_COMPONENT)                                                               list(APPEND CMAKE_ABSOLUTE_DESTINATION_FILES                                                                                                            "/opt/code/freecad-git/installs/mar22.01.22.linkdaily/FreeCAD.app/Contents/libexec/import site site.addsitedir('/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages')/lib/python3.9')/import site;/opt/code/freecad-git/installs/mar22.01.22.linkdaily/FreeCAD.app/Contents/libexec/import site site.addsitedir('/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages')/lib/python3.9')/site-packages')")
  if(CMAKE_WARN_ON_ABSOLUTE_INSTALL_DESTINATION)
    message(WARNING "ABSOLUTE path INSTALL DESTINATION : ${CMAKE_ABSOLUTE_DESTINATION_FILES}")
  endif()
  if(CMAKE_ERROR_ON_ABSOLUTE_INSTALL_DESTINATION)                                                                                                          message(FATAL_ERROR "ABSOLUTE path INSTALL DESTINATION forbidden (by caller): ${CMAKE_ABSOLUTE_DESTINATION_FILES}")                                  endif()                                                                                                                                                file(INSTALL DESTINATION "/opt/code/freecad-git/installs/mar22.01.22.linkdaily/FreeCAD.app/Contents/libexec/import site site.addsitedir('/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages')/lib/python3.9')" TYPE DIRECTORY FILES
    "/opt/code/freecad-git/freecad-src/src/MacAppBundle/import site"                                                                                       "/opt/code/freecad-git/freecad-src/src/MacAppBundle/ site.addsitedir('/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages')"
    )
endif()
```

i can obviously go hacking into the `CMakeLists.txt` file, but prefer to get this working in a _less hack way_ if possible.

and thanks for making such awesome/ambitious fork of freecad, looking forward to publish a release based on your _linkdaily_ branch.

ipatch, 2022-03-24:
**update**

got the same results using either of the two examples for the,

```
-DPYTHON_INCLUDE_DIR=
```

realthunder, 2022-03-24:
Can you please post the full `cmake` command output, with both stdout and stderr capture? I use roughly the same [cmake command](https://github.com/realthunder/FreeCADMakeImage/blob/14054e8eeb61330756e62993e4b96d67ec6221d7/mkimg.sh#L828) as yours to build my release, but I am not that familiar with MacOS.

PS. I use a [custom branch](https://github.com/realthunder/coin) of Coin3D for my release, which contains some additional visual features. I use `brew edit` to build a local package for that. It is not compulsory though, as the official Coin3D can still be used.

ipatch, 2022-03-27:
a little follow up to my 2nd comment above, the below error message line,

```
"/opt/code/freecad-git/freecad-src/src/MacAppBundle/import site": No such
```

is related to the homebrew specific `.pth` file related to pybind11, specifically

```
$(brew --prefix)/Cellar/pybind11/2.9.1/lib/python3.9/site-packages/homebrew-pybind11.pth
```

which contains

```
import site; site.addsitedir('/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages')
```

i believe the current `CMakeLists.txt` is unable to process the `import site;` as it is not run through the python interpreter. the hacky fix is to update the contents of that file to,

```
/Users/brewmaster/homebrew/Cellar/pybind11/2.9.1/libexec/lib/python3.9/site-packages
```

obviously that path needs to be updated for the specific `$USER` homebrew prefix ie. `$(brew --prefix)`

luzpaz, 2022-04-08:
Anymore progress with this?

ipatch, 2022-05-16:
been doing some work on an updated build...

i believe if you add the below lines to your mac app bundling `CMakeLists.txt` it should prevent the error i describing above. i have tested this locally on my machine.

https://github.com/FreeCAD/FreeCAD/blob/cd53c06b9899b03fe88d2a2fe8258422e7ac9c09/src/MacAppBundle/CMakeLists.txt#L35-L38

luzpaz, 2022-05-16:
Nice!