realthunder/FreeCAD #300: Segfault when extruding/padding

zoopp, 2022-04-02:
I'm modeling a cap and trying to make a small extrusion on a concentric face to act as a sort of lip. Rather than explaining with words, I'll attach a video showing what I was doing:

https://user-images.githubusercontent.com/1645023/161400395-cf24c296-f09a-4285-96e7-efcc29b3c5e2.mp4

[Project_File.zip](https://github.com/realthunder/FreeCAD/files/8403345/Project_File.zip)

---
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.20.30919 (Git)
Build type: Release
Branch: makepkg
Hash: 879b0b989ced7a40dcc0381caa9329f635d0b169
Python version: 3.10.4
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

luzpaz, 2022-07-15:
@zoopp can't reproduce on 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.712.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.07.12-edge)
Hash: 8ce366738c70dde14b276f68f6e2f6a380eac004
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
```

Can you try latest LinkDaily or Snap edge ?

zoopp, 2022-07-16:
Hey @luzpaz, I can confirm that with the latest LinkDaily the crash is gone.

Just out of curiosity though, I would have expected that padding the extrusion would pad inward thus creating a small circular "lip". Am I misunderstanding things or is this a separate issue?

https://user-images.githubusercontent.com/1645023/179350507-dbc8db77-2152-4bbb-a8b1-5c8e8c1584c7.mp4

realthunder, 2022-07-16:
You are using the wrong tool. There is no native tool in `PartDesign` can do this. But you can use the `PartDesign Warp` feature to wrap any non `PartDesign` feature inside a body. Make sure you activate your body first. Then, go to `Part` workbench, and create an `3D Offset` feature based on the `Extrusion`.

https://user-images.githubusercontent.com/1207888/179357403-3041cbb9-0ba9-4ae2-9a4b-3d8a8d4e5c28.mp4

luzpaz, 2022-07-16:
Closing ticket

closed 2022-07-16 by luzpaz