realthunder/FreeCAD #301: Running all Test WB on LinkDaily is problematic

luzpaz, 2022-04-03:
@realthunder is there a way to tell TestWB (when running in the GUI) to automatically say yes to any recompute requests for opening unit test files?
I recommend not running tests in the GUI until you figure that out (because you're potentially pressing the enter key hundreds of times!). I'm running LinkDaily from console now to see how it fairs. 

Here are the results of running `freecad-realthunder.cmd -t 0`
```shell
======================================================================
ERROR: testCTRIA3Element (MeshTestsApp.NastranReader)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Mesh/MeshTestsApp.py", line 583, in testCTRIA3Element
    m = Mesh.read(f"{self.test_dir}/NASTRAN_Test_GRID_CTRIA3.bdf")
RuntimeError: File does not exist: /snap/freecad-realthunder/14/usr/Mod/Mesh/App/TestData/NASTRAN_Test_GRID_CTRIA3.bdf

======================================================================
ERROR: testDelimitedGRIDElement (MeshTestsApp.NastranReader)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Mesh/MeshTestsApp.py", line 573, in testDelimitedGRIDElement
    m = Mesh.read(f"{self.test_dir}/NASTRAN_Test_Delimited_GRID_CQUAD4.bdf")
RuntimeError: File does not exist: /snap/freecad-realthunder/14/usr/Mod/Mesh/App/TestData/NASTRAN_Test_Delimited_GRID_CQUAD4.bdf

======================================================================
ERROR: testEightCharGRIDElement (MeshTestsApp.NastranReader)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Mesh/MeshTestsApp.py", line 568, in testEightCharGRIDElement
    m = Mesh.read(f"{self.test_dir}/NASTRAN_Test_GRID_CQUAD4.bdf")
RuntimeError: File does not exist: /snap/freecad-realthunder/14/usr/Mod/Mesh/App/TestData/NASTRAN_Test_GRID_CQUAD4.bdf

======================================================================
ERROR: testSixteenCharGRIDElement (MeshTestsApp.NastranReader)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Mesh/MeshTestsApp.py", line 578, in testSixteenCharGRIDElement
    m = Mesh.read(f"{self.test_dir}/NASTRAN_Test_GRIDSTAR_CQUAD4.bdf")
RuntimeError: File does not exist: /snap/freecad-realthunder/14/usr/Mod/Mesh/App/TestData/NASTRAN_Test_GRIDSTAR_CQUAD4.bdf

======================================================================
ERROR: test00 (PathTests.TestPathStock.TestPathStock)
Test CreateBox
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Path/PathTests/TestPathStock.py", line 48, in setUp
    self.job.addProperty('App::PropertyLink', 'Proxy')
RuntimeError: Property TestPathStock#Job.Proxy already exists

======================================================================
ERROR: test01 (PathTests.TestPathStock.TestPathStock)
Test CreateCylinder
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Path/PathTests/TestPathStock.py", line 48, in setUp
    self.job.addProperty('App::PropertyLink', 'Proxy')
RuntimeError: Property TestPathStock1#Job.Proxy already exists

======================================================================
ERROR: test10 (PathTests.TestPathStock.TestPathStock)
Verify FromTemplate box creation.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Path/PathTests/TestPathStock.py", line 48, in setUp
    self.job.addProperty('App::PropertyLink', 'Proxy')
RuntimeError: Property TestPathStock2#Job.Proxy already exists

======================================================================
ERROR: test11 (PathTests.TestPathStock.TestPathStock)
Verify FromTemplate cylinder creation.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/snap/freecad-realthunder/14/usr/Mod/Path/PathTests/TestPathStock.py", line 48, in setUp
[... 103 lines cut ...]
  File "/snap/freecad-realthunder/14/usr/Mod/Test/Document.py", line 1813, in testDocument
    self.assertEqual(self.Obs.signal.pop(), 'DocAbortTransaction')
AssertionError: 'ObjChanged' != 'DocAbortTransaction'
- ObjChanged
+ DocAbortTransaction

----------------------------------------------------------------------
Ran 588 tests in 73.295s

FAILED (failures=3, errors=10)
```

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.402.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.02-edge)
Hash: b126e1545ee0a35aa3c9831058a384e1cc1f88fc
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
````

realthunder, 2022-04-03:
You can turn off the prompt with the following option

[image]

luzpaz, 2022-04-03:
Thanks! Maybe this should be automatically by default be temporarily enabled when only running TestWB? Maybe that's overkill. Just thinking out loud.

realthunder, 2022-04-03:
Yes, there should be something like that. I'll add it later.

luzpaz, 2022-04-03:
Thanks for your time.

closed 2022-04-03 by luzpaz