realthunder/FreeCAD #304: SIGSEGV, Segmentation fault

VincentFrangi, 2022-04-06:
Hello,

Every time I try to move a rectangle with draft move tool a `SIGSEGV, Segmentation fault` occur

```python
OS: Manjaro Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.20.Unknown
Build type: Release
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: C/Default (C)
```

```python
FreeCAD 0.20, Libs: 0.20RUnknown
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

----------------
GuiCommand: Move
[(<Part::Part2DObject>, 'Face1')]
Pick start point
----------------
GuiCommand: Move
[(<Part::Part2DObject>, 'Face1')]
Pick start point
Pick end point
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x42560) [0x7f1868e35560]
#1  0x7f186b8f44b5 in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /usr/lib/libCoin.so.80+0x535
#2  0x7f186ce95beb in Gui::SoFCUnifiedSelection::GLRenderInPath(SoGLRenderAction*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x5b
#3  0x7f186b8f44c5 in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /usr/lib/libCoin.so.80+0x545
#4  0x7f186b6ce9d7 in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x67
#5  0x7f186ceb7adb in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x8b
#6  0x7f186b6cf352 in SoAction::apply(SoPathList const&, int) from /usr/lib/libCoin.so.80+0x5b2
#7  0x7f186ceb5816 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x96
#8  0x7f186b6d5ab7 in SoGLRenderActionP::renderSingle(SoNode*) from /usr/lib/libCoin.so.80+0x277
#9  0x7f186b6d5fbe in SoGLRenderActionP::render(SoNode*) from /usr/lib/libCoin.so.80+0x11e
#10  0x7f186ceb7adb in Gui::SoBoxSelectionRenderAction::beginTraversal(SoNode*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x8b
#11  0x7f186b6cf6ba in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0x2ea
#12  0x7f186ceb5b19 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x29
#13  0x7f186b9c8952 in SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) from /usr/lib/libCoin.so.80+0xa2
#14  0x7f186b9c8b07 in SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) from /usr/lib/libCoin.so.80+0xc7
#15  0x7f186b9c913c in SoRenderManager::render(SoGLRenderAction*, int, int, int) from /usr/lib/libCoin.so.80+0x9c
#16  0x7f186cf8e943 in Gui::View3DInventorViewer::renderScene() from /usr/lib/freecad/lib/libFreeCADGui.so+0x393
#17  0x7f186cf4171d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x30d
#18  0x7f186cf4b2a2 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x42
#19  0x7f186acc84dc in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x99c
#20  0x7f186ad794c3 in QFrame::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x23
#21  0x7f186a21b32a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0xaa
#22  0x7f186ac921b5 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x75
#23  0x7f186caf30e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x88
#24  0x7f186a21c5aa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x13a
#25  0x7f186acbc1fc in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/libQt5Widgets.so.5+0x3c
#26  0x7f186acbd592 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0xd52
#27  0x7f186acbe075 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x505
#28  0x7f186acbcac3 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x283
#29  0x7f186acbe075 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x505
#30  0x7f186acbcac3 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x283
[... 9 lines cut ...]
#40  0x7f186acbde96 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x326
#41  0x7f186acbde96 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x326
#42  0x7f186acbcac3 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /usr/lib/libQt5Widgets.so.5+0x283
#43  /usr/lib/libQt5Widgets.so.5(+0x184b90) [0x7f186ac9db90]
#44  0x7f186acc8855 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0xd15
#45  0x7f186ac921c6 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x86
#46  0x7f186caf30e8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x88
#47  0x7f186a21c5aa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x13a
#48  0x7f186a21d0a9 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt5Core.so.5+0x199
#49  /usr/lib/libQt5Core.so.5(+0x2d4678) [0x7f186a264678]
#50  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x193) [0x7f1867cc0163]
#51  /usr/lib/libglib-2.0.so.0(+0xab9e9) [0x7f1867d169e9]
#52  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7f1867cbd6c5]
#53  0x7f186a26857a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6a
#54  0x7f186a21488b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12b
#55  0x7f186a21ffd7 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x97
#56  0x7f186ca435da in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x41a
#57  freecad(+0x4be4) [0x5567efedbbe4]
#58  /usr/lib/libc.so.6(+0x2d310) [0x7f1868e20310]
#59  /usr/lib/libc.so.6(__libc_start_main+0x81) [0x7f1868e203c1]
#60  freecad(+0x4f75) [0x5567efedbf75]
```

Thanks in advance,
Vincent

luzpaz, 2022-04-08:
@VincentFrangi did you compile FreeCAD yourself? 
Are you sure you're using the Link branch ?

VincentFrangi, 2022-04-08:
> @VincentFrangi did you compile FreeCAD yourself? Are you sure you're using the Link branch ?

I cannot compile the freecad-linkstage-3 (arch package) anymore due to third party lib issues..
Instead I'm using this package:
[freecad-linkstage3-bin](https://aur.archlinux.org/packages/freecad-linkstage3-bin)

luzpaz, 2022-04-08:
Can you test the LinkDaily AppImage? https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11#Assets  
Builidng FC on Arch/AUR has been problematic for a long time now. Also LinkDaily snap just started acting up as well. So AppImage is the most reliable at this moment

VincentFrangi, 2022-05-24:
I've tried the snap package and I didn't encountered any issues !
Seems to be way more stable than the aur package