realthunder/FreeCAD #305: snap version for Ubuntu no access to files

imhavoc-classic, 2022-04-06:
The snap version of RT Freecad seems to be running in a sandbox or something. I have no access to my file system from the Open dialog.

realthunder, 2022-04-07:
Yes, that's the expected behavior. You have access to your home directory. And you can create symlink to other place in your home directory.

imhavoc-classic, 2022-04-07:
Thanks for the prompt response!

So, do I have to `ln -s` every file I want to work on into the home directory?

realthunder, 2022-04-11:
Yes, looks like so. @ppd Any better idea?

ppd, 2022-04-11:
There's the `removable-media` interface, too: https://snapcraft.io/docs/removable-media-interface

> removable-media allows read/write access to mounted removable storage in /media, /run/media and /mnt.
> 
> For example, this interface can be used to provide access to mounted USB sticks and external hard drives from the desktop environment.

You can connect the interface via the software store or via the cmd line:

```console
$ snap connect freecad-realthunder:removable-media
```

Normally, this is handled by using desktop portals: https://github.com/flatpak/xdg-desktop-portal
Those allow access (by means of proxying) to almost any file on the system. FreeCAD, however, has trouble with using those portals as hasn't been designed (at least so far) with their implications in mind. Therefore, these portals are disabled in the freecad-realthunder snap.

I'm guessing your files are on some network device? In this case, mounting them according to the `removable-media` interface specifications and possibly soft linking them for easier access would be my solution.