realthunder/FreeCAD #307: snap version shows graphic bugs

wohltat, 2022-04-07:
The snap version is completely unusable for me because the GUI has major graphic issues on my system.
[image]

The gui elements that i can see seem to be shifted and partially disappear when i try to interact with them.
[image]

--------------------
Linux Mint XFCE
> uname -a
Linux computername 5.4.0-100-generic #113-Ubuntu SMP Thu Feb 3 18:43:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

> lsb_release -a 
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20.1
Release:	20.1
Codename:	ulyssa

wohltat, 2022-04-07:
i tried to remove settings (`.FreeCAD` and `config/FreeCAD`) and remove snap gtk themes but nothing improved the situation.

wohltat, 2022-04-07:
when i start the snap package via command line i get this:
``` bash
> env BAMF_DESKTOP_FILE_HINT=/var/lib/snapd/desktop/applications/freecad-realthunder_freecad-realthunder.desktop /snap/bin/freecad-realthunder
FreeCAD 0.20, Libs: 2022.406R14555 (Git shallow)
Fatal Error at file "/home/username/snap/freecad-realthunder/common/.FreeCAD/link.system.cfg", line 1, column 1
   Message: invalid document structure
Malformed Parameter document: Root group not found
Malformed Parameter document: Root group not found in file /home/username/snap/freecad-realthunder/common/.FreeCAD/link.system.cfg.
Continue with an empty configuration.
Fatal Error at file "/home/username/snap/freecad-realthunder/common/.FreeCAD/link.user.cfg", line 1, column 1
   Message: invalid document structure
Malformed Parameter document: Root group not found
Malformed Parameter document: Root group not found in file /home/username/snap/freecad-realthunder/common/.FreeCAD/link.user.cfg.
Continue with an empty configuration.
Adding snap-specific PYTHONPATH to sys.path: /home/username/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/16/lib/python3.8/site-packages:/snap/freecad-realthunder/16/usr/lib/python3/dist-packages
5.038791 <asm3.main> init_gui.py(14): no solver backend found
connect failed: No such file or directory
```

I don't know how it should look like, but there is a `Root` FCParamGroup.
``` bash
> cat snap/freecad-realthunder/common/.FreeCAD/link.system.cfg                             
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>

  <FCParamGroup Name="Root">
    <FCParamGroup Name="Modules">
      <FCParamGroup Name="Image">
        <FCText Name="HelpIndex">Image/Help/index.html</FCText>
        <FCText Name="WorkBenchName">Image</FCText>
        <FCText Name="WorkBenchModule">ImageWorkbench.py</FCText>
      </FCParamGroup>
      <FCParamGroup Name="Path">
        <FCText Name="HelpIndex">Path/Help/index.html</FCText>
        <FCText Name="WorkBenchName">Path</FCText>
        <FCText Name="WorkBenchModule">PathWorkbench.py</FCText>
      </FCParamGroup>
      <FCParamGroup Name="Spreadsheet">
        <FCText Name="HelpIndex">Spreadsheet/Help/index.html</FCText>
        <FCText Name="WorkBenchName">Spreadsheet</FCText>
        <FCText Name="WorkBenchModule">SpreadsheetWorkbench.py</FCText>
      </FCParamGroup>
    </FCParamGroup>
  </FCParamGroup>

</FCParameters>
```

luzpaz, 2022-04-08:
I'm running 
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.07-edge)
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
```

snap stores the UserParms in a separate place. In the FreeCAD python console type:

```python
FreeCAD.ConfigGet("UserAppData")
FreeCAD.ConfigGet("UserParameter")
FreeCAD.ConfigGet("SystemParameter")
```

```python
>>> FreeCAD.ConfigGet("UserAppData")
'/home/foo/snap/freecad-realthunder/common/.FreeCAD/'
>>> FreeCAD.ConfigGet("UserParameter")
'/home/foo/snap/freecad-realthunder/common/.FreeCAD/link.user.cfg'
>>> FreeCAD.ConfigGet("SystemParameter")
'/home/foo/snap/freecad-realthunder/common/.FreeCAD/link.system.cfg'
```

Backup the `link.user.cfg' and `link.system.cfg` files and restart FreeCAD again

wohltat, 2022-04-08:
I deleted the whole snap/freecad-realthunder directory. Some console error are gone but the main graphics issue persists.
``` bash
> env BAMF_DESKTOP_FILE_HINT=/var/lib/snapd/desktop/applications/freecad-realthunder_freecad-realthunder.desktop /snap/bin/freecad-realthunder
FreeCAD 0.20, Libs: 2022.407R14555 (Git shallow)
Adding snap-specific PYTHONPATH to sys.path: /home/boogieman/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/17/lib/python3.8/site-packages:/snap/freecad-realthunder/17/usr/lib/python3/dist-packages
3.541031 <asm3.main> init_gui.py(14): no solver backend found
connect failed: No such file or directory
```

luzpaz, 2022-04-08:
JFYI, the `connect failed: No such file or directory` can be ignored. It's related to libspnav ([fixed in upstream v0.3](https://github.com/FreeSpacenav/libspnav/releases/tag/v0.3))

The `3.541031 <asm3.main> init_gui.py(14): no solver backend found` is weird. That should of been fixed

luzpaz, 2022-04-08:
I was trying to install @studiopetrikas's MidnightUI in the Snap edge, and it started showing instability for me as well. The UI starts to hang and then is unresponsive. I can't tell if it was mucking around with MidnightUI or perhaps there were issues that wohltat was experiencing and I didn't see them (I don't use Link super often and haven't used this latest version at all. 

```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.07-edge)
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
```

luzpaz, 2022-04-13:
@ppd do you mind checking for any of the graphic anomalies in the freecad-realthunder edge ?

ppd, 2022-04-13:
Very interesting, I see no such issues on Intel UHD620/Mesa/Gnome 42. @wohltat what GPU driver do you use?
There is some weirdness going on with recent Qt versions regarding transparency etc. (see https://github.com/realthunder/FreeCAD_assembly3/issues/1173#issuecomment-1086866519)

I suspect the issues are a combination of those "fresh" Qt packages + the compositor + maybe the GPU driver.

ppd, 2022-04-13:
Aha, I'm blind: https://forum.snapcraft.io/t/empty-dropdown-menus/29392

There's a very recent thread about empty dropdowns on AMD hardware. Do you have an AMD GPU?

wohltat, 2022-05-20:
Still have the same bugs in the latest snap.

@ppd I have an NVidia card in my ubuntu/mint machine.
```
>>> inxi -G  
Graphics:  Device-1: NVIDIA GF108M [GeForce GT 620M/630M/635M/640M LE] driver: nvidia v: 390.151 
           Display: x11 server: X.Org 1.20.13 driver: nvidia unloaded: fbdev,modesetting,nouveau,vesa 
           resolution: 1366x768~60Hz, 1920x1080~60Hz 
           OpenGL: renderer: GeForce GT 630M/PCIe/SSE2 v: 4.6.0 NVIDIA 390.151 

>>> compiz --version
Compiz 0.9.14.1
```

I tried the snap on another system (arch linux). And there it runs fine, except that it does not use my settings. But i guess this is normal behavior.

ppd, 2022-05-20:
@wohltat Same driver version on both installations? Where did you get the driver from, the Ubuntu repos? Could you try the nouveau driver on the Mint installation?

wohltat, 2022-05-29:
@ppd 
> Same driver version on both installations?

No. It is an Intel Graphics on my other machine. I guess it is not necessary to post further info on the working system.

> Where did you get the driver from, the Ubuntu repos? Could you try the nouveau driver on the Mint installation?

Yes Ubuntu repos as far as i can remember. Proprietary Nvidia drivers.

luzpaz, 2022-07-15:
@wohltat can you still reproduce on latest snap edge ?

wohltat, 2022-09-06:
I tested it today and i have the same behavior on the latest snap (2022.08.20)

wohltat, 2023-03-31:
still the same on 2023.03.21