realthunder/FreeCAD #309: Potential regression: "Illegal storage access" when adding fillet to arc edge

CharlemagneLasse, 2022-04-08:
I've noticed that FreeCAD sometimes cannot add a fillet to an edge. Even when it can do this perfectly fine on the opposite site of a symmetric design. While doing some tests with the official build, it was seen that:

* 0.19.4 uses a lot of clock cycles and the program crashes
* 0.20 just refuses the creation of the fillet

I've posted the minimal design example + some screenshots at https://forum.freecadweb.org/viewtopic.php?p=587189

During this test, I've also used one of your most recent releases from https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11

Interestingly, it caused a different kind of crash. First it showed a "Illegal storage access" and the whole program crashes when I select the next edge or when I close the fillet tool and try again.

If I correctly interpret the answer from [drmacro](https://forum.freecadweb.org/viewtopic.php?p=587189#p587189), then it seems to work with a newer OCC. But I am unsure if this user actually tested the correct thing - the image doesn't look like it.

But we at least know that it just fails (without crashing the whole program) with an older version of freecad/OCC. There were also some similar discussion where someone mentioned that it [might be fixed with OCC 7.6.1](https://github.com/FreeCAD/FreeCAD/issues/6625#issuecomment-1084908697). But I don't know I should get a freecad linkstage build with OCC 7.6.1 to test it.

CharlemagneLasse, 2022-04-09:
I cannot reproduce this crash with DRAWEXE (like it can be done with [#6625](https://github.com/FreeCAD/FreeCAD/issues/6625)). So it is is most likely an actual regression with your FreeCAD on windows with OCCT 7.6.0

realthunder, 2022-04-11:
I have tested the file in your referenced post, using the exact same release version of OCC 7.6.1 with my branch and upstream on Linux produce the same hard crash at [ChFi3d_Builder::PerformIntersectionAtEnd()](https://github.com/Open-Cascade-SAS/OCCT/blob/d2abb6d844231cb8f29be6894440874a4700e4a5/src/ChFi3d/ChFi3d_Builder_C1.cxx#L1581). Using debug build of OCC throws an exception instead of hard crash because of extra boundary check, and the exact crash site is at [this line](https://github.com/Open-Cascade-SAS/OCCT/blob/d2abb6d844231cb8f29be6894440874a4700e4a5/src/ChFi3d/ChFi3d_Builder_C1.cxx#L1694). The reason of crash is because the index is out of range of sequence `SeqFil`.

The fact of no hard crash in earlier version does not imply that there is no problem there. Because the index is out of boundary, it may be writing into some other unrelated but non-critical variable.

blobfish, 2022-04-11:
@CharlemagneLasse @realthunder @chennes
Has anyone ran their changes through the occt public regression tests? I am not a lover of DRAWEXE or TCL, but I have ran the regression tests a few times and it is pretty smooth.

Where do you guys plan to land your changes? If you guys are on board with the occt CLA and plan to submit your changes to them, you can exclude me from the conversation. I don't want any changes on my branch that are also added to the official branch. I am trying to eliminate conflicts as I rebase my branch from one official release to the next.

If you are on matrix and want to chat, I set up an occt room a while back.
https://matrix.to/#/#occt:matrix.org

chennes, 2022-04-11:
@blobfish I'm going to submit this upstream.

blobfish, 2022-04-11:
@chennes  ok thanks Chris.

CharlemagneLasse, 2022-04-13:
Bug was forwarded to https://tracker.dev.opencascade.org/view.php?id=32929