realthunder/FreeCAD #312: [Daily] Hidden-Line Pick Bugs - "Auto-Transparent Pick" disables when Hidden-Line pick is highlighted, and Hidden-Line pick does not work with "Experimental" Render Cache

scottmudge, 2022-04-11:
Reporting from the tip of LinkDaily as of 04/11/2022.

Two issues with Hidden-Line Pick:

1. The Auto-Transparent Pick feature successfully allows the user to select hidden lines when [Shift] is held, however the transparent preview disables itself when a hidden-line pick is being previewed. The selected parent object (if occluded) does get placed on-top **after** selection is made.

Example:

https://user-images.githubusercontent.com/19617165/162728867-31374e78-5d79-405a-843e-ccf035257ec1.mp4

2. You are probably aware, but the Hidden-Line Pick feature does not work with the Experimental render cache.

closed 2022-04-13 by realthunder

scottmudge, 2022-04-13:
Nice! Looks beautiful.

Thanks!

realthunder, 2022-04-13:
The above commit fixed a problem I found when investigating your report. This may or may not be the problem you have.

Please note that the `Auto Transparent Pick` is supposed to work in all render cache modes. But there are different behaviors. The one you showed in screen cast is the expected behavior of `Auto Transparent Pick` in non-experimental caching mode. It will make the pre-selected object transparent but will not make the object `show on top`, and hence will not enable auto hidden line picking. That's why you still need the `Shift` key to enable hidden line picking.

With experimental caching, pre-selected object will be shown on top, and hidden line picking is enabled without needing to hold the `Shift` key. The above commit fixed a bug in this mode.

scottmudge, 2022-04-13:
It seems to have fixed my problem, and the experimental render cache works now too. Perhaps I didn't describe the original problem well to go along with the video. In any case, it's fixed!

Unrelated, but what's the plan/trajectory for merging the primary FreeCAD repo into LinkDaily? There's a few features now I'd love to use (e.g., the offset feature in the sketcher), but I don't want to abandon the benefits of your branch either.

realthunder, 2022-04-13:
There is huge code refactoring in upstream recently, Sketcher WB being one of them. I'll do a merge once upstream 0.20 is released, probably in a month or so I think.

scottmudge, 2022-04-13:
Yeah it's impressive work. Sounds awesome, I look forward to it! Thanks!

commit 7c1a3c19 (2022-04-13): Gui: fix auto transparent pick

Fixes #312
commit 637dad52 (2022-04-26): Gui: fix auto transparent pick

Fixes #312