realthunder/FreeCAD #316: I cannot run Last Windows version (20220405)

DoumeDuSud, 2022-04-11:
I cannot run the last Windows version, I try all 3 : 

- FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220405
- FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64
- FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220405

The previous one is working (FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111)

The software doesn't appear in Task manager. 
When I start the `FreeCADCmd.exe` I have the following error : 
```
Error processing line 1 of C:\Users\dguiraud\Prog_Portable\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\bin\lib\site-packages\distutils-precedence.pth:

  Traceback (most recent call last):
    File "C:\Users\dguiraud\Prog_Portable\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\bin\lib\site.py", line 169, in addpackage
      exec(line)
    File "<string>", line 1, in <module>
    File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
    File "<frozen importlib._bootstrap>", line 982, in _find_and_load_unlocked
    File "<frozen importlib._bootstrap>", line 925, in _find_spec
    File "<frozen importlib._bootstrap_external>", line 1423, in find_spec
    File "<frozen importlib._bootstrap_external>", line 1395, in _get_spec
    File "<frozen importlib._bootstrap_external>", line 1526, in find_spec
    File "<frozen importlib._bootstrap_external>", line 1569, in _fill_cache
  OSError: [WinError 87] Paramètre incorrect: 'C:\\Users\\dguiraud\\Prog_Portable\\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\\bin'

Remainder of file ignored
Error in sitecustomize; set PYTHONVERBOSE for traceback:
OSError: [WinError 87] Paramètre incorrect: 'C:\\Users\\dguiraud\\Prog_Portable\\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\\bin'
Error in usercustomize; set PYTHONVERBOSE for traceback:
OSError: [WinError 87] Paramètre incorrect: 'C:\\Users\\dguiraud\\Prog_Portable\\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\\bin'
FreeCAD 0.20, Libs: 2022.405R26244 +4689 (Git)
Traceback (most recent call last):
  File "<string>", line 257, in <module>
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 982, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 925, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1423, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1395, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1526, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1569, in _fill_cache
<class 'OSError'>: [WinError 87] ParamÃ¨tre incorrect: 'C:\\Users\\dguiraud\\Prog_Portable\\FreeCAD-asm3-Daily-Win-Conda-Py3.9-20220405-x86_64\\bin'
[FreeCAD Console mode <Use Ctrl-Z plus Return to exit.>]
>>>
```

I don't know if I can capture some more info in a log

DoumeDuSud, 2022-06-09:
I have the same issue with le last 0.20 freecad release : #https://github.com/FreeCAD/FreeCAD/issues/7000

luzpaz, 2022-07-15:
@DoumeDuSud can you run https://github.com/realthunder/FreeCAD/releases/tag/2022.07.09-edge ?

luzpaz, 2022-10-07:
ping @DoumeDuSud