realthunder/FreeCAD #317: Linux AppImage crashes when rending an object in Shadow Draw Style

KarlReynolds, 2022-04-11:


KarlReynolds, 2022-04-26:
I can confirm this issue still exists with the latest FreeCAD_assembly3 branch release:
OS: Debian GNU/Linux 9 (stretch) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

Also from looking at the source code relating to your responses, I have found an alternative temporary workaround to prevent the program crashing.
Users can disable the display of the navigation cube `Edit>Preferences>Display>Navigation>Navigation cube [un-checked]`. (The number pad keys can be used to perform the same function as the Navigation cube, to rotate the viewpoint).
When the Navigation Cube is not displayed Anti-Aliasing can be switched back on without crashing the program.

luzpaz, 2022-07-06:
@KarlReynolds please retest on latest LinkDaily release? Can you also produce this on the Snap 'edge' package?

KarlReynolds, 2022-07-06:
Retested on latest LinkDaily release. (Unable to test Snap 'edge' package).
Results of LinkDaily test: Problem persists on latest build.

`Menu: Edit > Preferences > Display > Navigation > Navigation cube [checked] > Apply`

Object is displayed in Shadow Draw Style.
Program Segmentation Faults instantly as the Shadow Draw Style is chosen.

OS: Debian GNU/Linux 9 (stretch) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 2022.704.26244 +4971 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: a87320cc64699a8e4226bad2e9b37ac3a1296ab9
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)

Terminal output:
```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220704-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA2E8Sgi
FreeCAD 0.20, Libs: 2022.704R26244 +4971 (Git)
1.298491 <asm3.main> init_gui.py(14): no solver backend found
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7efd11a10060]
#1  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x4b71d6) [0x7efcff5f21d6]
#2  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x4b41c9) [0x7efcff5ef1c9]
#3  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x22c2c2) [0x7efcff3672c2]
#4  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x22c9eb) [0x7efcff3679eb]
#5  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x21c4f9) [0x7efcff3574f9]
#6  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x25d1aa) [0x7efcff3981aa]
#7  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x4ba29d) [0x7efcff5f529d]
#8  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0xc4b8e) [0x7efcff1ffb8e]
#9  0x7efd141e4f78 in NaviCubeShared::drawNaviCube(SoCamera*, bool, int, bool) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x1578
#10  0x7efd141e5316 in NaviCubeImplementation::drawNaviCube(bool) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x46
#11  0x7efd141aefed in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x4ed
#12  0x7efd1415d99d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x31d
#13  0x7efd14163d25 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x45
#14  0x7efd129b598d in QWidget::event(QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#15  0x7efd12a1eded in QFrame::event(QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x27
#16  0x7efd12b99efc in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x456
#17  0x7efd11f2599b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Core.so.5+0xcd
#18  0x7efd1298a811 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#19  0x7efd1299034d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#20  0x7efd13d0c3b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x89
#21  0x7efd11f27f88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Core.so.5+0x86
#22  0x7efd129b0642 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x28
#23  0x7efd129b0d6f in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x707
#24  0x7efd129b14e1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#25  0x7efd129b11ee in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#26  0x7efd129b14e1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#27  0x7efd129b11ee in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#28  0x7efd129b14e1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#29  0x7efd129b11ee in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#30  0x7efd129b14e1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
[... cut ...]
#53  0x7efd11f2499b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Core.so.5+0x187
#54  0x7efd11f28811 in QCoreApplication::exec() from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libQt5Core.so.5+0xfd
#55  0x7efd13c7d71b in Gui::Application::runApplication() from /tmp/.mount_FreeCA2E8Sgi/usr/bin/../lib/libFreeCADGui.so+0x44b
#56  /tmp/.mount_FreeCA2E8Sgi/usr/bin/FreeCADLink(+0x4659) [0x55a80622d659]
#57  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7efd119fd2e1]
#58  /tmp/.mount_FreeCA2E8Sgi/usr/bin/FreeCADLink(+0x49e9) [0x55a80622d9e9]
```

luzpaz, 2022-07-06:
Can you install the `snap install freecad-realthunder --edge` and try to reproduce ?

KarlReynolds, 2022-07-07:
No. As mentioned prior "(Unable to test Snap 'edge' package)".
Snap requires systemd which this OS installation doesn't use.
`uname output: Linux...4.19.0-9-amd64 #1 SMP Debian 4.19.118-2~mx17+1 (2020-06-14) x86_64 GNU/Linux`