realthunder/FreeCAD #323: [Feature Request] Import TechDraw Extensions from upstream FreeCAD

ToniTen, 2022-04-11:


letssolvesolvedproblems, 2022-06-13:
Just bumping this issue. The extensions which were added in 0.20 of the main branch are basically requirements for making drawings professionally and efficiently. They'd be really nice to see incorporated.

luzpaz, 2022-07-24:
Not a small ask, especially while realthunder is working on porting Toponaming (see his [status update](https://forum.freecadweb.org/viewtopic.php?p=611902#p611902) 3 days ago on the forum)

kiltel, 2023-02-15:
Your version of Freecad is great.  However, I also have an urgent request to update TechDraw Workbench to the latest version.

Symbian9, 2023-02-25:
> I'll make a new release soon, and then try to do a merge with upstream.

Its year might be already but merging upstream TechDraw is really needed for *FreeCADLink*.

Especially, new TechDrawTools:

- https://github.com/FreeCAD/FreeCAD/pull/5344
- https://github.com/FreeCAD/FreeCAD/pull/5383

Key contributors to TechDraw:

- https://github.com/FreeCAD/FreeCAD/commits?author=edi271
- https://github.com/FreeCAD/FreeCAD/commits?author=WandererFan

commit a2cde695 (2020-10-16): Gui: fix ViewProviderGeoFeatureGroupExtension migration problem

Fixes #323