realthunder/FreeCAD #332: Toponaming: Multi-solid body with conf tables and A3

Kuzma30, 2022-04-11:
Some info. 
This is test file for assembly with multi-solid body (with configuration tables). It has 4 "virtual" configuration set.
It has 1 Element for Body defined. When I change set of parameters v1 to v2, v3, v4 sometimes Elements001 go to error state. In this moment no problem with Body.
[roll.zip](https://github.com/realthunder/FreeCAD/files/8464868/roll.zip)

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Ukrainian/Ukraine (uk_UA)

```
16:57:01  0.130421 <asm3.main> assembly.py(789): Failed to get geometry element from roll#Body.Revolution.;#56;:L;RVL;:H67f:7,F;:L;RVL;:H67f:18,F;:L;RVL;:H67f:2a,F.Revolution.?Face7
16:57:01  Traceback (most recent call last):
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\assembly.py", line 787, in execute
    info = self.getInfo(False)
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\assembly.py", line 1264, in getInfo
    return getElementInfo(self.getAssembly().getPartGroup(),
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\assembly.py", line 1796, in getElementInfo
    raise RuntimeError('Failed to get geometry element from '
<class 'RuntimeError'>: Failed to get geometry element from roll#Body.Revolution.;#56;:L;RVL;:H67f:7,F;:L;RVL;:H67f:18,F;:L;RVL;:H67f:2a,F.Revolution.?Face7
16:57:01  3665.06 <App> solver.py(376)|Document.cpp(4266): Failed to recompute roll#Element001: Failed to get geometry element from roll#Body.Revolution.;#56;:L;RVL;:H67f:7,F;:L;RVL;:H67f:18,F;:L;RVL;:H67f:2a,F.Revolution.?Face7
16:57:01  Recompute failed! Please check report view.
16:57:01  0.007002 <asm3.main> assembly.py(3918): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 535, in _catch
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\solver.py", line 412, in solve
    return _solve(*args,**kargs)
  File "C:\Users\kuzem\AppData\Roaming\FreeCAD\Mod\FreeCAD_assembly3\freecad\asm3\solver.py", line 377, in _solve
    raise RuntimeError(
RuntimeError: Failed to recompute roll#Assembly
```

realthunder, 2022-04-14:
The topo naming algorithm has its limit. It cannot automatically unambiguously identify the new face after parameter changes every time. It mostly works if the geometry of the referenced face stays the same while the other part of the model changes. It often fails if otherwise, which is what happens in your case.

There are a few ways to solve this problem. You can create a auxiliary sketch that import some vertex from your revolving sketch and create a circle for use by the assembly element.

Or, you can install the `Configuration` into the assembly (or make a higher level configuration that drives the body configuration), and make that bottom face as a parameter to bind to the `Element001.LinkObject` property. So with your given file, for configuration `v1` the value is `=(Revolution._ref, <<Face7>>)`. And for `v2`, it is `=(Revolution._ref, <<Face1>>)`. In other word, use the table to explicitly set the element reference for each configuration.

Kuzma30, 2022-04-14:
> Or, you can install the `Configuration` into the assembly (or make a higher level configuration that drives the body configuration), and make that bottom face as a parameter to bind to the [...]

Can you make small example?

realthunder, 2022-04-14:
Sure. But wait till I get a deeper look. I've found some thing wrong by testing with your file.

Kuzma30, 2022-04-14:
That this is a topological problem was my assumption. I have done some research. Open the Sketcher and change the options in the table. I watched how the name of the edges changes in the Sketcher. It did not change, but the error in the assembly appeared. Could it be something else?

realthunder, 2022-04-14:
The screen cast below shows a potential problem with your approach. Depending on the order you change the configuration, there is a chance for the sketch to become invalid. This is because some sketch constraint disregards of value signess which means there are often more than one solutions even for a fully constrained sketch. The signess problem usually occur if you have sudden jump in constraint value which is exactly what I did in the screen cast. It would be better to decompose this sketch into several smaller sketch for better stability.

https://user-images.githubusercontent.com/1207888/163490844-3753fc69-57dd-47a5-9c7a-cf6a630ea560.mp4

I am still investigating why in some cases the face index will change even if the sketch looks fine. For example, the file in its state you uploaded, `Element001` is referencing `Revolution.Face7`. But when switching configuration from `v1` to `v2`, the same face becomes `Revolution.Face1`. The change causes topo naming encoding to change as well. It is possible to trace history through topo naming to find the same source edge in sketch, but that's not always reliable.