realthunder/FreeCAD #333: [Bug] Pad result differs from preview, it's croocked

YosypC, 2022-04-11:
I'm having loads of problems with the latest release, bugs and crashes, one of which is this:

[image]

[image]

[1911MagHolsterBetter.zip](https://github.com/realthunder/FreeCAD/files/8466655/1911MagHolsterBetter.zip)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/Europe (en_150)

bambukouk, 2022-04-11:
It's the Draft001 that is causing the problem.
Delete it and the Pocket006 is behaving fine.

YosypC, 2022-04-11:
Thank you, but... I need that.

bambukouk, 2022-04-11:
I am not suggesting you don't need that...
but perhaps doing things in different order?
or in a different way? (eg boolean cut? instead of draft?)

YosypC, 2022-04-11:
I guess I'm going to post it to the main FC page, then.

Reordering the features would take me 20 minutes of re-adjusting lost references and what not, it would be the only solution until a fix is proposed, but the point of noticing the developer of bugs is for him to spare users from workarounds.

I've tried a boolean cut but the resulting model is also bugged.

Thanks for discovering the cause, it's really important.

luzpaz, 2022-04-13:
@YosypC did you post to the FC forum? What's the URL ?

bambukouk, 2022-04-13:
https://forum.freecadweb.org/viewtopic.php?f=8&t=67941
and there is also an answer to the same question:
https://github.com/realthunder/FreeCAD/issues/334

luzpaz, 2022-04-13:
Per @kisolre in the forum:
> Draft creates a broken model. Pad after that is not independent. It is created on top of Draft, which is already broken so Pad is also broken. Draft shows multiple errors because of Refine=True. There are some errors even without refine. Also when I open the file in master there is no source for the latest Pad.

YosypC, 2022-04-13:
> @YosypC did you post to the FC forum? What's the URL ?

I also posted a new Issue on the Master FC, as already highlighted by the platform.