realthunder/FreeCAD #343: [Feature Request] - Fine adjustment with mouse wheel mode to accompany coarse adjustment with mouse wheel when Ctrl is held.

scottmudge, 2022-04-13:
The current behavior when adjusting data fields in the property view is that, when using the mouse wheel, the standard increment is by +/- 1.0 units. If [Ctrl] is held, that increases to +/- 10.0 units.

It would be nice if holding [Shift], or perhaps [Ctrl+Shift] while using the mouse wheel in this manner would reduce the increment to +/- 0.1 units.

If this is already an option, I apologies for the duplicate. I could not find a pre-existing setting.

bambukouk, 2022-04-13:
I didn't know about use of Ctrl to control (pardon the pun) sensitivity of mouse wheel inputs.
Thank you :+1:  I have learned something today
As for another keyboard input to make it more sensitive (instead of Ctrl making it less) - yes, please :+1: 

but before that, a question .....
there are dozens upon dozens of fine enhancements like this, 
that an average user misses most of the time 
(unless he/she is hardcore keyboard entry fanatic who has memorised them all...)

Is it possible to have some hint about available keyboard input variations? when using mouse?

scottmudge, 2022-04-13:
I'm not aware of any on-screen hints. I believe you can check all of the existing keybindings in the keyboard UI, but quirky stuff like this is unlikely to be documented. I kinda just figure it out by accident.

realthunder, 2022-04-13:
Right click inside any (well, most of) spin box widget, you can change the `Single step` to specify the resolution of wheel stepping. This setting is persistent to some extent.

scottmudge, 2022-04-14:
Good to know. But I actually like the single step. I use it most frequently. But occasionally I like to visually adjust 0.1 mm at a time, so a key binding for that would be nice. I also like the coarse 10mm adjustment w/ Ctrl held though, so I don't think I'd override that.

It's alright, I can look for where the Ctrl+ scroll behavior is in the code and add one for ctrl+shift.

scottmudge, 2022-04-14:
I can't find the behavior coded anywhere, it looks like it is a default behavior from Qt:

> When the user triggers a step whilst holding the [Qt::ControlModifier](https://doc.qt.io/qt-5/qt.html#KeyboardModifier-enum), QAbstractSpinBox steps by 10 instead of making a single step. This step modifier affects wheel events, key events and interaction with the spinbox buttons. Note that on macOS, Control corresponds to the Command key.

wohltat, 2022-04-14:
I like the idea of a second shortcut for smaller value changes.

Here is another idea. What about relative value changes?
Most of the time you are not interested in increasing a value by 1 or 0.1 but instead you want to make it like 10% or 1% bigger.
Combined with some "snapping algorithm" this would fit most needs i think. With snapping algorithm i mean that you can also reach round values and not ending up with odd decimal places everywhere.

An idea to solve that would be to have a mix between normal / linear and relative / multiplicative behavior.
The behavior would be normal in a way, but the "active" decimal place changes depending on the value.

kevenwyld, 2022-04-17:
I also would appreciate a shortcut for fine adjust.  In my experience I've had more need to selectively move small increments than large ones with most of the spin boxes.

I also like the idea of adding it to the tool tip if possible as @bambukouk suggested, but that might be a lot more work.

Even a mention at the bottom of this dialog under the single step box would be better if the tooltip is too difficult to modify.  Though the tool tip would be more consistent with other FreeCAD shortcuts.
[image]