realthunder/FreeCAD #345: Measurement label invisible (w/ experimental renderer)

OxygenLiu, 2022-04-14:
[image]

```
OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

Screenshot at https://ibb.co/F4wg3dq

_Mod Edit: Added screenshot to the ticket directly for posterity_

bambukouk, 2022-04-14:
[image]

scottmudge, 2022-04-14:
Well there we go, that answers that.