realthunder/FreeCAD #348: xkbcommon: ERROR: failed to add default include path

OxygenLiu, 2022-04-15:
When launching Link Daily, the following error occurs, 

> xkbcommon: ERROR: failed to add default include path ././././././././././././/../share/X11/xkb

Although everything works fine, I am wondering what is the possible side effect, and how to avoid the error message.

```
OS: Pop!_OS 20.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

luzpaz, 2022-07-04:
@OxygenLiu can you still reproduce?

OxygenLiu, 2022-07-05:
seems all right with the latest 22020704 release.

closed 2022-07-05 by OxygenLiu